basler/pypylon #1: Unable to get color images using ImageFormatConverter

rezaxdi, 2018-05-01:
hi there,

I'm unable to run ImageFormatConverter on GrabResult : 

`NotImplementedError: Wrong number or type of arguments for overloaded function 'ImageFormatConverter_Convert'.
  Possible C/C++ prototypes are:
    Pylon::CImageFormatConverter::Convert(Pylon::IReusableImage &,Pylon::IImage const &)
    Pylon::CImageFormatConverter::Convert(Pylon::IReusableImage &,Pylon::CGrabResultPtr const &)
`

Is this function still incomplete ?

basler-oss, 2018-05-02:
Hi rezaxdi,

After looking into the source code I would actually expect that this function is implemented. Could you send me a copy of you code sample which triggers this issue? Thanks!

stefanklug, 2018-05-30:
@dmirecki We are working on a fix. The updates will be pushed in a few days.

LucasVandroux, 2018-05-31:
I just started to use pypylon and I am looking forward to be able to use the ImageFormatConverter.

stefanklug, 2018-06-06:
@dmirecki  I just pushed new changes to master. The ImageFormat converter should work now.

closed 2018-06-06 by stefanklug