basler/pypylon #2: Convert GrabImage to be used with OpenCV

LucasVandroux, 2018-05-31:
Hello,

I am trying to capture color images from my acA1920-40gc and to process them using OpenCV. According to #1 this ability should be soon implemented. Am I correct? Moreover, will the procedure be somehow similar to the one describe in [Getting Started with pylon and OpenCV](https://www.baslerweb.com/fp-1476182890/media/downloads/documents/application_notes/AW00136801000_Getting_Started_with_pylon5_and_OpenCV.pdf)?

Thanks a lot for your time.

Have a good day,

basler-oss, 2018-05-31:
Hi Lucas,

the mentioned document refers to our C++ API. On our community platform ImagingHub you'll find a [few examples which explains how to use pypylon](https://imaginghub.com/projects/97-face-detection-on-raspberry-pi#documentation) along with opencv. I hope thi helps.

All best

LucasVandroux, 2018-05-31:
I love your documentation! But I have two problems that I think are kinda related. First, the`instant_camera.PixelFormat = "RGB8"` doesn't work with my Ace acA1920-40gc. I guess it doesn't support this `PixelFormat` but I have no idea where to find the precise list of the `PixelFormat` supported by the camera.

Then actually when I manage to save a picture, the picture is looking weird and I don't know how to change this:
[image]

However when I upload it to GitHub, it looks perfectly normal:
[image]