basler/pypylon #3: Find List PixelFormat for Specific Camera

LucasVandroux, 2018-05-31:
Hello,

As mentioned in #2 I am having trouble capturing images with my Ace acA1920-40gc. My biggest problem is that the camera is a color one but I can't figure out how to get a color image and all the images are in black and white.

Following the instruction on [ImagingHub](https://imaginghub.com/projects/97-face-detection-on-raspberry-pi#documentation), I figured out that I should change the `PixelFormat` of the camera before starting the recording:
```python
# Simply get the first available pylon device.
first_device = pylon.TlFactory.GetInstance().CreateFirstDevice()
camera = pylon.InstantCamera(first_device)
camera.Open()

# Optional if you set it in Pylon Viewer
camera.PixelFormat = 'RGB8'

camera.StartGrabbing(pylon.GrabStrategy_LatestImages)
```

However, I get the following error:
```
An exception occurred.
Traceback (most recent call last):
  File "demo1.py", line 111, in detect_defects
    camera.PixelFormat = 'RGB8'
  File "/home/lucas/anaconda3/envs/demo1/lib/python3.6/site-packages/pypylon/pylon.py", line 4199, in __setattr__
    self.GetNodeMap().GetNode(attribute).SetValue(val)
  File "/home/lucas/anaconda3/envs/demo1/lib/python3.6/site-packages/pypylon/genicam.py", line 2611, in SetValue
    return _genicam.IEnumeration_SetValue(self, entry)
_genicam.InvalidArgumentException: Feature 'PixelFormat' : cannot convert value 'RGB8', the value is invalid. : InvalidArgumentException thrown in node 'PixelFormat' while calling 'PixelFormat.FromString()' (file 'Enumeration.cpp', line 132)
```

My guess is that `RGB8` is definitely not the right thing to use. Is there a list of the different `PixelFormat` I can use with my camera?

Thanks in advance for your time.

Have a good day,

basler-oss, 2018-06-01:
Hi Lucas, the list of supported pixel formats can be found in the [camera manual](https://docs.baslerweb.com/#t=en%2Fpixel_format.htm&rhsearch=acA1920-40gc) or just open the camera with the pylon viewer and check for the list of pixel formats in the feature tree. Your camera has the following formats: 
Mono 8, Bayer RG 8, Bayer RG 12, Bayer RG 12 Packed, YUV 4:2:2 Packed and YUV 4:2:2 (YUYV) Packed
--

LucasVandroux, 2018-06-01:
@basler-oss thank you very much, this was really the problem. Now I know where to find the information.

I just used the following line instead of the previous one:

`camera.PixelFormat = 'BayerRG8'`

However, the problem was not in the fact the camera was not recording in color but it was that I was not saving the picture in the right format. Indeed, I should first convert it to the RGB format:

`img = cv2.cvtColor(grabResult.Array, cv2.COLOR_BAYER_RG2RGB)`

Thanks a lot for your patience and support :+1:

closed 2018-06-01 by LucasVandroux

zarzar31, 2021-11-22:
Thank you for your code. I also captured 16 bits BAYERRG images, and converted them to RGB using cv2.cvtColor like your code.
But, what the problem is that, my captured frames seems reddish. 
So, could I get some idea to revise them?