basler/pypylon #4: Activating Automatic Image Adjustment using pypylon

LucasVandroux, 2018-06-01:
Hello,
I really like the `Automatic Image Adjustment` feature from the Pylon Viewer:

[image]

Is there a way to activate the same feature directly from pypylon? I looked quickly in the examples but I didn't find anything.

Thanks a lot for your time,

- Lucas

basler-oss, 2018-06-04:
@LucasVandroux: This function is currently not available in our APIs. We have individual API calls for activating auto brightness, auto white ballance auto gain. You might use these functions instead.

closed 2018-06-09 by LucasVandroux

paaraujo, 2019-02-03:
Dear @basler-oss , I am starting with the pypylon, how can I activate the auto brightness, auto white ballance and auto gain parameters?

weberhen, 2019-05-13:
@paaraujo 

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreatteFirstDevice())
camera.Open()
camera.GainAuto.SetValue("Once") sets the auto gain for example. 

To have a list of all parameters you can change #40