basler/pypylon #5: no released binary file in the link

alwynchen123, 2018-06-04:
Hi,

I didn't find the compiled wheel file in the link. I guess the readme.md is not updated?

Thanks,

Alwyn

basler-oss, 2018-06-04:
That's true. We temporarily removed the binaries as a bug needs to be fixed (see [issue #1](https://github.com/basler/pypylon/issues/1)). We put them back in the next days.

MonkeyKH, 2018-06-12:
Do you Have a Presumed time scale for when you want to add the binaries again? 
thanks

stefanklug, 2018-06-14:
@MonkeyKH We are currently working on automating the binary releases. First release candidates are currently flowing into the releases page. Feel free to test them. I expect the final release to be ready next week.

MonkeyKH, 2018-06-15:
Thanks a lot For all of your guys hard work, i will definetly do that.

closed 2018-07-12 by stefanklug