basler/pypylon #6: OS X Compiling support

LauriTK, 2018-06-04:
Hi, is there plans for OS X support?

basler-oss, 2018-06-05:
No, we do not plan to add support for macOS. However - as we release the pypylon also in source you might try to port pypylon to macOS by yourself. Any input/feedback concerning this approach would be highly welcome :-)

LauriTK, 2018-06-05:
Hi, thank you for fast reply. I tried, but had some problems with OS X permissions and anaconda python 3.6 with the OS X's Library folder where all the libs are. Will try it again after doing some research with swig and OS X compiling.

basler-oss, 2018-06-05:
Great - keep us posted about your progress :-)

matt-phair, 2018-06-29:
I've been trying to nail this down and am running into some problems.

* `BinPath` and the conditional in `make()` are missing `macosx-10.9-x86_64`. I've added it.
* `call_pylon_config()` ends up putting a `-n ` at the front of the stuff it returns. I've manually stripped that out.
* The `PYLON_ROOT` isn't defined correctly. I couldn't find it on OSX after installing Pylon, but I found them packaged up in the osx download
* `-Wl,--enable-new-dtags` isn't accepted by osx's `ld`. I've removed it from the `ExtraLinkArgs`
* `-Wl,-E` also isn't accepted by `ld`. I've removed it after the `self.call_pylon_config("--libs")` call

After these changes, `pip install .` seems to work correctly. However when call `from pypylon import pylon` it errors out:

```
Python 3.6.5 (v3.6.5:f59c0932b4, Mar 28 2018, 05:52:31)
[GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from pypylon import pylon
Traceback (most recent call last):
  File "/Users/mphair/programming/pypylon/pypylon/pylon.py", line 42, in swig_import_helper
    return importlib.import_module(mname)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'pypylon._pylon'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/mphair/programming/pypylon/pypylon/pylon.py", line 45, in <module>
    _pylon = swig_import_helper()
  File "/Users/mphair/programming/pypylon/pypylon/pylon.py", line 44, in swig_import_helper
    return importlib.import_module('_pylon')
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_pylon'
>>>
```

Thoughts?

matt-phair, 2018-07-02:
I'm starting to think that maybe the issue is that the latest OSX install for the Pylon SDK is 5.0.5 but the latest for other platforms is 5.0.12.

@basler-oss Do you think this is causing the build issues. If so, can you provide a later osx build for the SDK?

stefanklug, 2018-07-06:
@matt-phair It is pretty unlikely that the version (5.0.5) is causing these problems. I suspect your way of linking the pylon framework might be a problem here. I didn't have a deeper look, but did you link pylon as shown in the OSX samples using the -framework switch?
If you can provide a branch with your changes in the setup.py maybe I can have a look at it.

matt-phair, 2018-07-06:
@stefanklug Still working on getting something non-messy to put up so others can see. However, during the refactor I found that I was setting `PylonConfig` in `setup.py` to be an executable built on a different platform. The only `pylon-config` executable I found after the OSX installation was in `/Library/Frameworks/pylon.framework/Versions/A/Resources/Tools/pylon-config` and that doesn't come with the commands necessary for the `setup.py` script to run.

```
$ /Library/Frameworks/pylon.framework/Versions/A/Resources/Tools/pylon-config --libdir
Unknown command --libdir
Usage: /Library/Frameworks/pylon.framework/Versions/A/Resources/Tools/pylon-config <command>... [library]

Return metainformation about the given library:
    pylon               the pylon C++ library (default)

  Allowed Commands:
    -h, --help          Show this help message
    --version           Prints the full pylon version
    --version-major     Prints the major version
    --version-minor     Prints the minor version
    --version-subminor  Prints the subminor version
    --version-build     Prints the build version
```

matt-phair, 2018-07-06:
I used the linux `pylon-config` results to hard-code some of the paths for OSX. I also had to change all of the `%include <pylon/*>` lines in the `*.i` files to be `%include <*>` but now it compiles and imports. However, it doesn't find a camera.

```
>>> camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pypylon/pylon.py", line 1025, in CreateFirstDevice
    return _pylon.TlFactory_CreateFirstDevice(self, *args)
_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 917)
```

If I run the pylon viewer app the camera shows up.

matt-phair, 2018-07-06:
And with that search query in mind, I found this:

https://github.com/mabl/PyPylon/issues/18#issuecomment-253078532

which works great. Looks like their conclusion is that @basler-oss needs to update their OSX release.

matt-phair, 2018-07-06:
For the lazy:

Before you execute python call this:

`export LD_LIBRARY_PATH=/Library/Frameworks/pylon.framework/Libraries`

matt-phair, 2018-07-06:
I created a pull request for this: https://github.com/basler/pypylon/pull/13 but it needs some work before merging.

matt-phair, 2018-07-10:
@stefanklug You mentioned maybe being able to look at the code I'm using. I'm not sure how to finish this off without changing the `.i` files. If you can have a look and maybe lead me to the right answer, that would be great. #13

stefanklug, 2018-07-13:
The issue with the .i files is that swig doesn't support OS X frameworks. On OS X if you link to the framework "pylon" an include <pylon/something.h> automatically resolves to /path/to/pylon.framework/Headers/something.h
See also http://swig.10945.n7.nabble.com/SWIG-and-Mac-OSX-frameworks-td11602.html

There is no clean way around this problem without fixing swig. A workaround (not tested) might be  to recreate that structure with symlinks. Something like:
```
mkdir <build_dir>/includes
cd <build_dir>/includes
ln -s /path/to/pylon.framework/Headers pylon
#call swig with -I<build_dir>/includes
```

matt-phair, 2018-07-13:
@stefanklug Thank! That worked and I've updated #13 to not have all of the `.i` file changes.

lauritk, 2018-08-01:
Great work @matt-phair! I will test this soon as possible. Something wrong with the merge?

matt-phair, 2018-08-01:
>Something wrong with the merge?

Not sure what you mean. The PR can merge cleanly. It's failing one check, but all of the PRs on the repo are failing that check, even ones that only change the help file. I'm not sure what that check is or how to fix it.

stefanklug, 2018-08-06:
The PR looks mostly good to me. Don't worry about the failing checks. The history needs some cleanup though and it should be rabased.
One question remains: Is the LD_LIBRARY_PATH issue still there? This should be fixed before merging the changes. According to one of our OS X devs, the libs should work fine. I suspect a rpath issue in the _pylon.so and the _genicam.so
See http://jorgen.tjer.no/post/2014/05/20/dt-rpath-ld-and-at-rpath-dyld/ for more details.

matt-phair, 2018-08-06:
>One question remains: Is the LD_LIBRARY_PATH issue still there?

Yeah, still there.

matt-phair, 2018-08-06:
@stefanklug I've rebased. Its the first time I've worked outside of the standard branch and merge paradigm so I'm not sure if what I did was correct. I ended up rebasing and squashing onto my forked master and then making a new PR from that. You can see it here #29

AlanHorman, 2018-08-31:
I always receive the error 
> Traceback (most recent call last):
>   File "C:/Users/rwilhelm/Desktop/pypylon/samples/grabmultiplecameras.py", line 56, in <module>
>     for i, cam in enumerate(cameras):
>   File "C:\Users\rwilhelm\AppData\Local\Continuum\Anaconda3\lib\site-packages\pypylon\pylon.py", line 1860, in __getitem__
>     if args[0] >= self.GetSize():
> NameError: name 'args' is not defined

even if I copy and paste the entire code for grabbing with multiple cameras.

Why is it not working?

I'm using Anaconda with Python 3.5 on Windows 7.

matt-phair, 2018-08-31:
@AlanHorman This is the issue specifically for compiling on OSX. Maybe you meant to start a new issue?

nimamox, 2019-02-19:
Has there been any updates on using PyPylon on macOS?

Knoose, 2019-02-26:
@nimamox, I was able to built it on my Mac and have been using it. I used the following branch and just followed the instructions in the Readme. https://github.com/basler/pypylon/tree/feature/python37

I had to modify a few lines of code but it does work.

gmnamra, 2019-05-13:
The link is no longer valid. And I do not see a binary wheel for maxOS ( OS X ). Any pointers ?

Knoose, 2019-05-13:
The master branch now supports the ability to build a binary wheel for macOSX. This was added in Release 1.4.0 and can be found from the release notes (https://github.com/Basler/pypylon/releases).

harshakasi, 2019-06-01:
Hello,

I have downloaded the latest pypylon Release 1.4.0 package. Tried my best to build the code on Mac OSx and failed. Following is the output I get from executing this command: `sudo python setup.py develop`

The result of this command was the following:

> fatal: not a git repository (or any of the parent directories): .git
`WARNING:root:git not found or invalid tag found.
WARNING:root:-> Building version from date!
Building version: 2019.6.1.dev50815
call /usr/local/bin/swig -python -outdir /Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon -o /Users/harshakasi/Downloads/pypylon-1.4.0 2/generated/genicam_wrap.cpp -c++ -Wextra -Wall -threads -modern -DSWIG_PYTHON_LEGACY_BOOL -I/Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes -I/Library/Frameworks/pylon.framework/Headers/GenICam -I/Library/Frameworks/pylon.framework/Headers -I/Library/Frameworks/pylon.framework/Headers/GenICam genicam.i
Deprecated command line option: -modern. This option is now always on.
NodeCallback.i:166: Warning 401: Nothing known about base class 'CNodeCallback'. Ignored.
IPort.i:23: Warning 403: Class 'GENAPI_NAMESPACE::IPort' might be abstract, no constructors generated,
/Library/Frameworks/pylon.framework/Headers/GenICam/GenApi/IBase.h:57: Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
PortImpl.i:52: Warning 403: Class 'GENAPI_NAMESPACE::CPortImpl' might be abstract, no constructors generated,
/Library/Frameworks/pylon.framework/Headers/GenICam/GenApi/IBase.h:57: Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
call /usr/local/bin/swig -python -outdir /Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon -o /Users/harshakasi/Downloads/pypylon-1.4.0 2/generated/pylon_wrap.cpp -c++ -Wextra -Wall -threads -modern -DSWIG_PYTHON_LEGACY_BOOL -I/Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes -I/Library/Frameworks/pylon.framework/Headers/GenICam -I/Library/Frameworks/pylon.framework/Headers -I/Library/Frameworks/pylon.framework/Headers/GenICam -I/Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes -I/Library/Frameworks/pylon.framework/Headers/GenICam -I/Library/Frameworks/pylon.framework/Headers -I/Library/Frameworks/pylon.framework/Headers/GenICam pylon.i
Deprecated command line option: -modern. This option is now always on.
/Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes/pylon/_ImageFormatConverterParams.h:47: Warning 322: Redundant redeclaration of 'GCC_DIAGNOSTIC_AWARE',
/Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes/pylon/_InstantCameraParams.h:47: Warning 322: previous declaration of 'GCC_DIAGNOSTIC_AWARE'.
../genicam/NodeCallback.i:166: Warning 401: Nothing known about base class 'CNodeCallback'. Ignored.
../genicam/IPort.i:23: Warning 403: Class 'GENAPI_NAMESPACE::IPort' might be abstract, no constructors generated,
/Library/Frameworks/pylon.framework/Headers/GenICam/GenApi/IBase.h:57: Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
../genicam/PortImpl.i:52: Warning 403: Class 'GENAPI_NAMESPACE::CPortImpl' might be abstract, no constructors generated,
/Library/Frameworks/pylon.framework/Headers/GenICam/GenApi/IBase.h:57: Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
/Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes/pylon/PylonImageBase.h:38: Warning 403: Class 'Pylon::CPylonImageBase' might be abstract, no constructors generated,
/Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes/pylon/ReusableImage.h:50: Warning 403: Method Pylon::IReusableImage::IsSupportedPixelType(Pylon::EPixelType) const might not be implemented.
/Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes/pylon/WaitObject.h:223: Warning 503: Can't wrap 'operator struct _WaitObjectPosix_t*' unless renamed to a valid identifier.
/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'long_description_content_type'
  warnings.warn(msg)
running develop
Checking .pth file support in /Library/Python/2.7/site-packages/
/usr/bin/python -E -c pass
TEST PASSED: /Library/Python/2.7/site-packages/ appears to support .pth files
running egg_info
writing pypylon.egg-info/PKG-INFO
writing top-level names to pypylon.egg-info/top_level.txt
writing dependency_links to pypylon.egg-info/dependency_links.txt
reading manifest file 'pypylon.egg-info/SOURCES.txt'
writing manifest file 'pypylon.egg-info/SOURCES.txt'
running build_ext
building 'pypylon._genicam' extension
cc -fno-strict-aliasing -fno-common -dynamic -g -Os -pipe -fno-common -fno-strict-aliasing -fwrapv -DENABLE_DTRACE -DMACOSX -DNDEBUG -Wall -Wstrict-prototypes -Wshorten-64-to-32 -DNDEBUG -g -fwrapv -Os -Wall -Wstrict-prototypes -DENABLE_DTRACE -arch i386 -arch x86_64 -pipe -DSWIG_TYPE_TABLE=pylon -I./src/genicam -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c /Users/harshakasi/Downloads/pypylon-1.4.0 2/generated/genicam_wrap.cpp -o build/temp.macosx-10.14-intel-2.7/Users/harshakasi/Downloads/pypylon-1.4.0 2/generated/genicam_wrap.o -Wno-unknown-pragmas -fPIC -g0 -Wall -O3 -Wno-switch -I/Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes -I/Library/Frameworks/pylon.framework/Headers/GenICam
In file included from /Users/harshakasi/Downloads/pypylon-1.4.0 2/generated/genicam_wrap.cpp:4164:
[... cut ...]
/Users/harshakasi/Downloads/pypylon-1.4.0 2/generated/genicam_wrap.cpp:45394:46: warning: conversion from string literal to 'char *' is deprecated [-Wc++11-compat-deprecated-writable-strings]
  pDynamicCastException = PyErr_NewException(SWIG_name ".DynamicCastException", pGenericException, NULL);
                                             ^
/Users/harshakasi/Downloads/pypylon-1.4.0 2/generated/genicam_wrap.cpp:3285:22: note: expanded from macro 'SWIG_name'
#define SWIG_name    "_genicam"
                     ^
12 warnings and 1 error generated.
error: command 'cc' failed with exit status 1
`

harshakasi, 2019-06-01:
What could be going wrong ?
Please help

harshakasi, 2019-06-01:
> @nimamox, I was able to built it on my Mac and have been using it. I used the following branch and just followed the instructions in the Readme. https://github.com/basler/pypylon/tree/feature/python37
> 
> I had to modify a few lines of code but it does work.

@Knoose Would you stil recommend to use this build ?

gmnamra, 2019-06-01:
Good news

harshakasi, 2019-06-03:
> Hello,
>
> I have downloaded the latest pypylon Release 1.4.0 package. Tried my best to build the code on Mac OSx and failed. Following is the output I get from executing this command: `sudo python [...]

Hi folks,

My colleague was able to help me out just by browsing to the part in setup.py --> under the class BuildSupportLinux(BuildSupport):
1. adjusting the path of my pylon.framework path related strings by modifying 
class BuildSupportMacOS(BuildSupport):

    FrameworkPath = **'/Library/Frameworks/'**

    FrameworkName = 'pylon.framework'

    PylonConfig = os.path.join(FrameworkPath, FrameworkName, '**Versions/A/Resources/Tools/pylon-config.sh**')
2. and just below that adding compiler flags accordingly:

ExtraCompileArgs = [
                       '-Wno-unknown-pragmas',
                       '-fPIC',
                       '-g0',
                       '-Wall',
                       '-O3',
                       '-Wno-switch',
                       **'-stdlib=libc++'**
                       ]

    ExtraLinkArgs = [
                    '-g0',
                    '-Wl,-rpath,@loader_path',
                    '-Wl,-framework,pylon',
                    '-F' + FrameworkPath,
                    **'-stdlib=libc++'**
                    ]

Voila!. Works like magic. Thanks for contributing one and all folks.

rarunsrivatsan, 2019-06-04:
Sorry, I made the changes as suggested above, but it still throws the following error:
clang: error: invalid deployment target for -stdlib=libc++ (requires OS X 10.7 or later)
error: command 'gcc' failed with exit status 1

I am currently running it in OS X 10.14.5, so I dont know why it is an issue. Can someone help me?

Knoose, 2019-06-04:
Do you have the XCode Developer Tools installed to your machine?

rarunsrivatsan, 2019-06-04:
Yes I have Xcode developers tools installed

Knoose, 2019-06-04:
Can you follow the instructions found under the **Installation from Source** heading in the readme? Make sure you have PCRE and SWIG installed. See if that works for you. If it does, then you can build the wheel file by specifying `python setup.py bdist_wheel `. This requires wheel to be installed to your python environment.

rarunsrivatsan, 2019-06-04:
Thank you very much. That worked!

Knoose, 2019-06-04:
I'm glad that worked :)

bora-uyumazturk, 2019-06-05:
Hi, I've done all of this but am still having trouble building on macOS. I'm running OS Sierra version 10.12.6. I keep getting the following error:

error: command 'cc' failed with exit status 1

Any ideas?

agosztolai, 2020-10-29:
> > Hello,
> > I have downloaded the latest pypylon Release 1.4.0 package. Tried my best to build the code on Mac OSx and failed. Following is the output I get from executing this command: `sudo python setup.py develop`
> > The result of this command was the following:
> > > fatal: not a git repository (or any of the parent directories): .git
> > > `WARNING:root:git not found or invalid tag found.
> > > WARNING:root:-> Building version from date!
> > > Building version: 2019.6.1.dev50815
> > > call /usr/local/bin/swig -python -outdir /Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon -o /Users/harshakasi/Downloads/pypylon-1.4.0 2/generated/genicam_wrap.cpp -c++ -Wextra -Wall -threads -modern -DSWIG_PYTHON_LEGACY_BOOL -I/Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes -I/Library/Frameworks/pylon.framework/Headers/GenICam -I/Library/Frameworks/pylon.framework/Headers -I/Library/Frameworks/pylon.framework/Headers/GenICam genicam.i
> > > Deprecated command line option: -modern. This option is now always on.
> > > NodeCallback.i:166: Warning 401: Nothing known about base class 'CNodeCallback'. Ignored.
> > > IPort.i:23: Warning 403: Class 'GENAPI_NAMESPACE::IPort' might be abstract, no constructors generated,
> > > /Library/Frameworks/pylon.framework/Headers/GenICam/GenApi/IBase.h:57: Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
> > > PortImpl.i:52: Warning 403: Class 'GENAPI_NAMESPACE::CPortImpl' might be abstract, no constructors generated,
> > > /Library/Frameworks/pylon.framework/Headers/GenICam/GenApi/IBase.h:57: Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
> > > call /usr/local/bin/swig -python -outdir /Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon -o /Users/harshakasi/Downloads/pypylon-1.4.0 2/generated/pylon_wrap.cpp -c++ -Wextra -Wall -threads -modern -DSWIG_PYTHON_LEGACY_BOOL -I/Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes -I/Library/Frameworks/pylon.framework/Headers/GenICam -I/Library/Frameworks/pylon.framework/Headers -I/Library/Frameworks/pylon.framework/Headers/GenICam -I/Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes -I/Library/Frameworks/pylon.framework/Headers/GenICam -I/Library/Frameworks/pylon.framework/Headers -I/Library/Frameworks/pylon.framework/Headers/GenICam pylon.i
> > > Deprecated command line option: -modern. This option is now always on.
> > > /Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes/pylon/_ImageFormatConverterParams.h:47: Warning 322: Redundant redeclaration of 'GCC_DIAGNOSTIC_AWARE',
> > > /Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes/pylon/_InstantCameraParams.h:47: Warning 322: previous declaration of 'GCC_DIAGNOSTIC_AWARE'.
> > > ../genicam/NodeCallback.i:166: Warning 401: Nothing known about base class 'CNodeCallback'. Ignored.
> > > ../genicam/IPort.i:23: Warning 403: Class 'GENAPI_NAMESPACE::IPort' might be abstract, no constructors generated,
> > > /Library/Frameworks/pylon.framework/Headers/GenICam/GenApi/IBase.h:57: Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
> > > ../genicam/PortImpl.i:52: Warning 403: Class 'GENAPI_NAMESPACE::CPortImpl' might be abstract, no constructors generated,
> > > /Library/Frameworks/pylon.framework/Headers/GenICam/GenApi/IBase.h:57: Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
> > > /Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes/pylon/PylonImageBase.h:38: Warning 403: Class 'Pylon::CPylonImageBase' might be abstract, no constructors generated,
> > > /Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes/pylon/ReusableImage.h:50: Warning 403: Method Pylon::IReusableImage::IsSupportedPixelType(Pylon::EPixelType) const might not be implemented.
> > > /Users/harshakasi/Downloads/pypylon-1.4.0 2/pypylon/includes/pylon/WaitObject.h:223: Warning 503: Can't wrap 'operator struct _WaitObjectPosix_t*' unless renamed to a valid identifier.
> > > /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'long_description_content_type'
> > > warnings.warn(msg)
> > > running develop
> > > Checking .pth file support in /Library/Python/2.7/site-packages/
> > > /usr/bin/python -E -c pass
> > > TEST PASSED: /Library/Python/2.7/site-packages/ appears to support .pth files
> > > running egg_info
> > > writing pypylon.egg-info/PKG-INFO
> > > writing top-level names to pypylon.egg-info/top_level.txt
> > > writing dependency_links to pypylon.egg-info/dependency_links.txt
> > > reading manifest file 'pypylon.egg-info/SOURCES.txt'
> > > writing manifest file 'pypylon.egg-info/SOURCES.txt'
> > > running build_ext
> > > building 'pypylon._genicam' extension
> > > cc -fno-strict-aliasing -fno-common -dynamic -g -Os -pipe -fno-common -fno-strict-aliasing -fwrapv -DENABLE_DTRACE -DMACOSX -DNDEBUG -Wall -Wstrict-prototypes -Wshorten-64-to-32 -DNDEBUG -g -fwrapv -Os -Wall -Wstrict-prototypes -DENABLE_DTRACE -arch i386 -arch x86_64 -pipe -DSWIG_TYPE_TABLE=pylon -I./src/genicam -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c /Users/harshakasi/Downloads/pypylon-1.4.0 2/generated/genicam_wrap.cpp -o build/temp.macosx-10.14-intel-2.7/Users/harshakasi/Downloads/pypylon-1.4.0 2/generated/genicam_wrap.o -Wno-unknown-pragmas -fPIC -g0 -Wall -O3 -Wno-switch -I/Users/harshakasi/Dow [... cut ...]
[... 81 lines cut ...]
>                 '-Wl,-framework,pylon',
>                 '-F' + FrameworkPath,
>                 **'-stdlib=libc++'**
>                 ]
> ```
> 
> Voila! Works like magic. Thanks for contributing @sparshturaga one and all folks.

I made these changes except for 

> ExtraCompileArgs = [
> '-Wno-unknown-pragmas',
> '-fPIC',
> '-g0',
> '-Wall',
> '-O3',
> '-Wno-switch',
> **'-stdlib=libc++'**
> ]
>
> ```
> ExtraLinkArgs = [
>                 '-g0', [...]

at everything installed ok

thiesmoeller, 2020-10-29:
Your modification was required on pypylon master?
What OS version ?
Can you provide a pull request ?

agosztolai, 2020-11-01:
I am running macOS 10.15.7. 

I have only modified the following entries in `setup.py`

```
FrameworkPath = '/Library/Frameworks/'
FrameworkName = 'pylon.framework'
PylonConfig = os.path.join(FrameworkPath, FrameworkName, 'Versions/A/Resources/Tools/pylon-config.sh')
```

lars-ek, 2021-01-12:
With the above modification to setup.py I can install pypylon without error-message on mac os 11.1.
But apon import, I get an error:
#>>> from pypylon import pylon
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/homebrew/lib/python3.9/site-packages/pypylon/pylon.py", line 40, in <module>
    from . import _pylon
ImportError: dlopen(/opt/homebrew/lib/python3.9/site-packages/pypylon/_pylon.cpython-39-darwin.so, 2): Symbol not found: __ZN5Pylon9CInfoBase16SetPropertyValueERKN29GenICam_3_1_Basler_pylon_v5_18gcstringES4_
  Referenced from: /opt/homebrew/lib/python3.9/site-packages/pypylon/_pylon.cpython-39-darwin.so
  Expected in: flat namespace
 in /opt/homebrew/lib/python3.9/site-packages/pypylon/_pylon.cpython-39-darwin.so

Anybody an idea? Thx.

thiesmoeller, 2021-02-23:
Beginning with release 1.7.1 we have binary releases on all platforms, that you can install via pip.

update your pip version before by calling `python -m pip install --upgrade pip`

closed 2021-02-23 by thiesmoeller