basler/pypylon #8: Picture not processed chronologically with grabmultiplecamera.py

LucasVandroux, 2018-06-12:
Hello,

I am using the code sample [grabmultiplecameras.py](https://github.com/basler/pypylon/blob/master/samples/grabmultiplecameras.py) with 3 GigE cameras. It works well as I can capture the images from the all three cameras but the pictures don't arrive in chronological order.

Is this something I should change in the software or is this due to the `Packet-Delay` parameter that I set to 5000 for each of the three cameras to have them work on the same switch.

Thanks a lot for your help,

thiesmoeller, 2018-06-15:
On an `InstantCameraArray ` `RetrieveResult `will work in a round-robin fashion on the grabbing cameras.
This could result in the re-ordering you observe. 

If you need strict chronological ordering you have to reorder them by yourself according to the Image timestamps.

`timestamp = grabResult.TimeStamp`

To let this work you have to synchronize the clocks of all cameras.
Depending on your camera model the IEEE1588 feature is available that enables this functionality [IEEE1588 on Basler GEV2.0 Cameras](https://docs.baslerweb.com/?filter=Camera:acA2440-20gc#t=en%2Fprecision_time_protocol.htm
).

LucasVandroux, 2018-06-20:
@thiesmoeller thanks a lot for the answer, this is clear now. I will use the timestamp trick to order my pictures.

P.S. For those, like me, who don't know what Round-Robin means, Wikipedia has the answer: [link](https://en.wikipedia.org/wiki/Round-robin_scheduling)

closed 2018-06-20 by LucasVandroux