basler/pypylon #9: How to grab image from multiple cameras synchronously

gachiemchiep, 2018-06-21:
Hello basler

Inside your samples directory there is a grabmultiplecameras.py file in which have the following comment:

```
    # Starts grabbing for all cameras starting with index 0. The grabbing
    # is started for one camera after the other. That's why the images of all
    # cameras are not taken at the same time.
    # However, a hardware trigger setup can be used to cause all cameras to grab images synchronously.
    # According to their default configuration, the cameras are
    # set up for free-running continuous acquisition.
```

My question is how to setup hardware trigger ? Could we use software trigger solution found in grabusinggrabloopthread.py for that ?

I tried using that but it lead to longer execution time than normal for loop as showed in grabmultiplecameras.py 

I paste parts of our code here. Hope you guys can help us.

How we init all camera

```

countOfImagesToGrab = 1
maxCamerasToUse = 2

# The exit code of the sample application.
exitCode = 0

# Init all camera
try :
    # Get the transport layer factory.
    tlFactory = pylon.TlFactory.GetInstance()

    # Get all attached devices and exit application if no device is found.
    devices = tlFactory.EnumerateDevices()
    if len(devices) == 0:
        raise pylon.RUNTIME_EXCEPTION("No camera present.")

    # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
    cameras = pylon.InstantCameraArray(min(len(devices), maxCamerasToUse))

    for i, camera in enumerate(cameras):
        camera.Attach(tlFactory.CreateDevice(devices[i]))
        
        camera.RegisterConfiguration(pylon.SoftwareTriggerConfiguration(), pylon.RegistrationMode_ReplaceAll,
                                     pylon.Cleanup_Delete)
        camera.RegisterImageEventHandler(ImageEventPrinter(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
        camera.GetDeviceInfo().SetPropertyValue("Expose time", '80')
        print("Expose time {}".format(camera.GetDeviceInfo().GetPropertyValue("Expose time")))
    
except genicam.GenericException as e:
    # Error handling
    print("An exception occurred. {}".format(e))
    exitCode = 1
```

Normal loop

```
now = datetime.datetime.now()

try:
    # Create and attach all Pylon Devices.
    for i, camera in enumerate(cameras):
        # Start grabbing
        camera.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)
        for i in range(countOfImagesToGrab):
            if not cameras.IsGrabbing():
                break
            grabResult = cameras.RetrieveResult(100, pylon.TimeoutHandling_ThrowException)
            img = grabResult.GetArray()
        
        # Stop grabbing
        camera.StopGrabbing()
        
except genicam.GenericException as e:
    # Error handling
    print("An exception occurred. {}".format(e))
    exitCode = 1

then = datetime.datetime.now()
execution_time = then - now

print("Execution time {} ms".format(execution_time.microseconds/1000))
```

```
Execution time 85 ms
```

grabusinggrabloopthread

```
now = datetime.datetime.now()

try:
    # Create and attach all Pylon Devices.
    for i, camera in enumerate(cameras):
        # Start grabbing
        camera.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)

        if camera.WaitForFrameTriggerReady(100, pylon.TimeoutHandling_ThrowException):
            grabResult = camera.ExecuteSoftwareTrigger();

        # Stop grabbing
        camera.StopGrabbing()
            
except genicam.GenericException as e:
    # Error handling
    print("An exception occurred. {}".format(e))
    exitCode = 1

then = datetime.datetime.now()
execution_time = then - now

print("Execution time {} ms".format(execution_time.microseconds/1000))
```

```
Execution time 157 ms
```

We want to use about 10 cameras, so is there any solution that we help us take 10 images in 200ms.

thiesmoeller, 2018-06-27:
SoftwareTrigger is always slower than an optimized HW trigger setup, due the fact that the PC is in the loop. Pylon has to wait in ```camera.WaitForFrameTriggerReady()```, transfer control back to python and send the software trigger command to the camera.

Please consult the camera documentation regarding hardware trigger on your device:
check https://docs.baslerweb.com/#t=en%2Ftriggered_image_acquisition.htm%23Hardware_Triggering&rhsearch=trigger

gachiemchiep, 2018-07-08:
@thiesmoeller 

Thank you man

closed 2018-07-19 by gachiemchiep

FadiAssad, 2023-01-31:
I can't see any use of the PTP for synchronisation, would it be possible to use it here?