basler/pypylon #10: LogicalErrorException on genicam call

dmirecki, 2018-06-27:
Hi,

When I started the callback.py from samples directory in the repository I gave an error in 18th line:

```
>>> genicam.Register(camera.GainRaw.Node, callback)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/dmirecki/miniconda3/envs/test/lib/python3.5/site-packages/pypylon/pylon.py", line 3768, in __getattr__
    return self.GetNodeMap().GetNode(attribute)
  File "/home/dmirecki/miniconda3/envs/test/lib/python3.5/site-packages/pypylon/genicam.py", line 1472, in GetNode
    return _genicam.INodeMap_GetNode(self, Name)
_genicam.LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16600)

```

Do you have any idea what went wrong?

thiesmoeller, 2018-06-27:
Assuming, you use a usb3vision camera:
please change the node-name from "GainRaw" to "Gain"
The name GainRaw is only available on GigEVision devices.

dmirecki, 2018-07-12:
Yes, it works, thanks for help

closed 2018-07-12 by dmirecki

satputeamit, 2018-08-10:
Hi,
when i call camera.open() it shows error.
code:
```
from pypylon import pylon
camera=pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.open()
```
Error:
>>> camera.open()
      Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
      File "/usr/local/lib/python3.5/dist-packages/pypylon/pylon.py", line 1832, in __getattr__
      return self.GetNodeMap().GetNode(attribute)
      File "/usr/local/lib/python3.5/dist-packages/pypylon/genicam.py", line 1250, in GetNode
      return _genicam.INodeMap_GetNode(self, Name)
      `pypylon.genicam.LogicalErrorException:` Node not existing

Can you help me?

thiesmoeller, 2018-08-12:
> File "", line 1, in

Hi,

The open has to be an uppercase Open.
( Relict from the C++ naming conventions used in pylon)

The error you see is the dynamic node lookup trying to find a node with the
name open

Thies

DilaraAlb, 2019-07-30:
> Assuming, you use a usb3vision camera:
> please change the node-name from "GainRaw" to "Gain"
> The name GainRaw is only available on GigEVision devices.

How to do that?

thiesmoeller, 2019-07-31:
remove '`Raw`' from the name of the feature:

`camera.GainRaw.SetValue(42)`

to:

`camera.Gain.SetValue(42)`

..

Resonanz, 2019-08-02:
Removing `Raw` does not solve it for me for my USB acA2040-90um. This also is an issue:

`AcquisitionFrameRateAbs`

```
---------------------------------------------------------------------------
LogicalErrorException                     Traceback (most recent call last)
<ipython-input-52-947f892c58cf> in <module>
      1 from pypylon_opencv_viewer import BaslerOpenCVViewer
      2 viewer = BaslerOpenCVViewer(camera)
----> 3 viewer.set_features(features)

C:\ProgramData\Anaconda3\lib\site-packages\pypylon_opencv_viewer\viewer.py in set_features(self, features)
    105             if feature_name is None:
    106                 raise ValueError("'name' attribute can't be None")
--> 107             pylon_feature = getattr(self._camera, feature_name)
    108 
    109             widget_kwargs['description'] = re.sub('([a-z])([A-Z])', r'\1 \2', feature_name) + " :"

C:\ProgramData\Anaconda3\lib\site-packages\pypylon\pylon.py in __getattr__(self, attribute)
   4037             return object.__getattr__(self, attribute)
   4038         else:
-> 4039             return self.GetNodeMap().GetNode(attribute)
   4040 
   4041     def __setattr__(self, attribute, val):

C:\ProgramData\Anaconda3\lib\site-packages\pypylon\genicam.py in GetNode(self, Name)
   1471 
   1472     def GetNode(self, Name: 'GENICAM_NAMESPACE::gcstring const &') -> "GENAPI_NAMESPACE::INode *":
-> 1473         return _genicam.INodeMap_GetNode(self, Name)
   1474 
   1475     def InvalidateNodes(self) -> "void":

LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16599)
```

thiesmoeller, 2019-08-02:
You own a Basler USB3Vision camera. This camera respects the SFNC2.0 standard regarding the names of all features.

The code, that you use is possibly an example for a GigEVision device. [ Which is SFNC 1.x ]

If you want to modify your code use below tables.

There is also the feature "migrationModeEnable" on the USB3 Transportlayer to do the conversion on the fly..

excerpt from pylon C++ manual:

Previous Parameter Name | SFNC 2.0 Equivalent | Parameter Type | Comments
-- | -- | -- | --
AcquisitionFrameCount | AcquisitionBurstFrameCount | Integer |  
AcquisitionFrameRateAbs | AcquisitionFrameRate | Float |  
AcquisitionStartEventFrameID | EventFrameBurstStartFrameID | Integer |  
AcquisitionStartEventTimestamp | EventFrameBurstStartTimestamp | Integer |  
AcquisitionStartOvertriggerEventFrameID | EventFrameBurstStartOvertriggerFrameID | Integer |  
AcquisitionStartOvertriggerEventTimestamp | EventFrameBurstStartOvertriggerTimestamp | Integer |  
AutoExposureTimeAbsLowerLimit | AutoExposureTimeLowerLimit | Float |  
AutoExposureTimeAbsUpperLimit | AutoExposureTimeUpperLimit | Float |  
AutoFunctionAOIUsageIntensity | AutoFunctionAOIUseBrightness | Boolean |  
AutoFunctionAOIUsageWhiteBalance | AutoFunctionAOIUseWhiteBalance | Boolean |  
AutoGainRawLowerLimit | Alias of AutoGainLowerLimit | Integer |  
AutoGainRawUpperLimit | Alias of AutoGainUpperLimit | Integer |  
AutoTargetValue | Alias of AutoTargetBrightness | Integer |  
BalanceRatioAbs | BalanceRatio | Float |  
BalanceRatioRaw | Alias of BalanceRatio | Integer |  
BlackLevelAbs | BlackLevel | Float |  
BlackLevelRaw | Alias of BlackLevel | Integer |  
ChunkExposureTimeRaw |   | Integer | ChunkExposureTimeRaw has been replaced with ChunkExposureTime. ChunkExposureTime is of type float.
ChunkFrameCounter |   | Integer | ChunkFrameCounter has been replaced with ChunkCounterSelector and ChunkCounterValue.
ChunkGainAll |   | Integer | ChunkGainAll has been replaced with ChunkGain. ChunkGain is of type float.
ColorAdjustmentEnable |   | Boolean | ColorAdjustmentEnable has been removed. The color adjustment is always enabled.
ColorAdjustmentHueRaw | Alias of ColorAdjustmentHue | Integer |  
ColorAdjustmentReset |   | Command | ColorAdjustmentReset has been removed.
ColorAdjustmentSaturationRaw | Alias of ColorAdjustmentSaturation | Integer |  
ColorTransformationValueRaw | Alias of ColorTransformationValue | Integer |  
DefaultSetSelector |   | Enumeration | See additional entries in UserSetSelector.
ExposureEndEventFrameID | EventExposureEndFrameID | Integer |  
ExposureEndEventTimestamp | EventExposureEndTimestamp | Integer |  
ExposureTimeAbs | ExposureTime | Float |  
ExposureTimeRaw | Alias of ExposureTime | Integer |  
FrameStartEventFrameID | EventFrameStartFrameID | Integer |  
FrameStartEventTimestamp | EventFrameStartTimestamp | Integer |  
FrameStartOvertriggerEventFrameID | EventFrameStartOvertriggerFrameID | Integer |  
FrameStartOvertriggerEventTimestamp | EventFrameStartOvertriggerTimestamp | Integer |  
GainAbs | Gain | Float |  
GainRaw | Alias of Gain | Integer |  
GammaEnable |   | Boolean | GammaEnable has been removed. Gamma is always enabled.
GammaSelector |   | Enumeration | The sRGB setting is automatically applied when LineSourcePreset is set to any other value than Off.
GlobalResetReleaseModeEnable |   | Boolean | GlobalResetReleaseModeEnable has been replaced with the enumeration ShutterMode.
LightSourceSelector | LightSourcePreset | Enumeration |  
LineDebouncerTimeAbs | LineDebouncerTime | Float |  
MinOutPulseWidthAbs | LineMinimumOutputPulseWidth | Float |  
MinOutPulseWidthRaw | Alias of LineMinimumOutputPulseWidth | Integer |  
ParameterSelector | RemoveParameterLimitSelector | Enumeration |  
ProcessedRawEnable |   | Boolean | ProcessedRawEnable has been removed because it is not needed anymore. The camera uses nondestructive Bayer demosaicing now.
ReadoutTimeAbs | SensorReadoutTime | Float |  
ResultingFrameRateAbs | ResultingFrameRate | Float |  
SequenceAddressBitSelector |   | Enumeration |  
SequenceAdvanceMode |   | Enumeration |  
SequenceAsyncAdvance |   | Command | Configure a asynchronous signal as trigger source of path 1.
SequenceAsyncRestart |   | Command | Configure a asynchronous signal as trigger source of path 0.
SequenceBitSource |   | Enumeration |  
SequenceControlConfig |   | Category |  
SequenceControlSelector |   | Enumeration |  
SequenceControlSource |   | Enumeration |  
SequenceCurrentSet | SequencerSetActive | Integer |  
SequenceEnable |   | Boolean | Replaced by SequencerConfigurationMode and SequencerMode.
SequenceSetExecutions |   | Integer |  
[... 35 lines cut ...]
LineSource | AcquisitionTriggerWait | FrameBurstTriggerWait |  
LineSource | UserOutput |   | Use UserOutput1, UserOutput2, or UserOutput3 etc. instead.
PixelFormat | BayerBG12Packed |   | The pixel  format BayerBG12p is provided by USB camera devices. The memory layout  of pixel format BayerBG12Packed and pixel format BayerBG12p is  different. See the camera User's Manuals for more information on pixel  formats.
PixelFormat | BayerGB12Packed |   | The pixel  format BayerGB12p is provided by USB camera devices. The memory layout  of pixel format BayerGB12Packed and pixel format BayerGB12p is  different. See the camera User's Manuals for more information on pixel  formats.
[... cut ...]
PixelFormat | RGB10Packed | RGB10 |  
PixelFormat | RGB12Packed | RGB12 |  
PixelFormat | RGB16Packed | RGB16 |  
PixelFormat | RGB8Packed | RGB8 |  
PixelFormat | RGBA8Packed | RGBa8 |  
PixelFormat | YUV411Packed | YCbCr411_8 |  
PixelFormat | YUV422_YUYV_Packed | YCbCr422_8 |  
PixelFormat | YUV444Packed | YCbCr8 |  
TestImageSelector | Testimage1 | GreyDiagonalSawtooth8 | GreyDiagonalSawtooth8 instead of Testimage1 is used for dart and pulse camera models.
TriggerSelector | AcquisitionStart | FrameBurstStart |