basler/pypylon #17: Segmentation fault (core dumped)

asavpatel92, 2018-07-10:
whenever I try to assign individual objects of InstantCameraArray into their own objects.
and then access object's properties or methods. Segmentation Fault occurs.

here's the example code to reproduce the error.

```
import logging
from pypylon import pylon
from utils.event_handlers import ConfigurationEventListener

MAX_CAMERAS = 2

formatter = logging.Formatter(
    '%(asctime)s:%(levelname)s:%(process)d:%(thread)d:%(module)s:%(funcName)s:%(lineno)s:%(message)s')
logger = logging.getLogger(__name__)

class Camera:

    def __init__(self, register_event_listener=False):
        tlFactory = pylon.TlFactory.GetInstance()
        # Get all attached devices and raise an error if no devices found
        devices = tlFactory.EnumerateDevices()
        if not devices:
            raise pylon.RUNTIME_EXCEPTION("No camera present.")

        # Create an array of instant cameras for the found devices and avoid
        # exceeding a maximum number of devices.
        cameras = pylon.InstantCameraArray(min(len(devices), MAX_CAMERAS))

        # Create and attach all Pylon Devices.
        for i, cam in enumerate(cameras):
            cam.Attach(tlFactory.CreateDevice(devices[i]))
            if register_event_listener:
                cam.RegisterConfiguration(ConfigurationEventListener(), pylon.RegistrationMode_Append,
                                          pylon.Cleanup_Delete)
            # Print the model name of the camera.
            logger.info("Initalizing device : {}".format(cam.GetDeviceInfo().GetModelName()))
            if "NIR" in cam.GetDeviceInfo().GetModelName().upper()[-3:]:
                self.nir = cam
            else:
                self.rgb = cam

if __name__ == '__main__':
    logging.basicConfig(level='DEBUG', format=formatter._fmt)
    cameras = Camera()
    
    # Segmentation fault (core dumped)
    print (cameras.nir.GetDeviceInfo().GetModelName())
```

MattsProjects, 2018-07-10:
Hello, I believe this is happening because the instant camera object cannot be copied (see InstantCamera.h) of the pylon SDK. So, accessing by cameras[0] is the way.

asavpatel92, 2018-07-11:
@MattsProjects thanks it works when I use it like you proposed.

closed 2018-07-11 by asavpatel92