basler/pypylon #20: Saving images grabbed from camera

glin123, 2018-07-16:
Hello;

I'm trying to take image from my new basler camera and save them to a certain folder. At the moment I have the camera taking pictures through a software trigger, I would like to eventually take in a sensor reading to trigger the camera to take a photo, if anyone have any info on this please let me know. But as for the images I can capture them but I'm not quite sure on how to save them. If anyone has anyone info on these 2 topics please let me know. Thank you.

matt-phair, 2018-07-16:
Here's how I do it:

```python
from PIL import Image
from pypylon import pylon

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
camera.PixelFormat = "Mono12"
camera.StartGrabbingMax(1)
try:
  while camera.IsGrabbing():
    result = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
    if result.GrabSucceeded():
      try:
        Image.fromarray(result.Array).save("/path/to/file.extension")
      finally:
        result.Release()
finally:
  camera.StopGrabbing()
  camera.close()
```

closed 2018-07-18 by glin123

cache91, 2018-08-01:
the above code doesnt seem to work

matt-phair, 2018-08-01:
>the above code doesnt seem to work

I didn't test it as the main important part I was trying to convey was the `Image.fromarray()` call. What problem are you running into with it?

cache91, 2018-08-02:
looks like the tiff format only works. Other formats fail

matt-phair, 2018-08-02:
Typo...should be `camera.Close()`