basler/pypylon #21: Camera trigger to capture image

glin123, 2018-07-16:
Hello:

I'm trying to get my camera to take an image every time time a sensor is triggered. At the moment my set up is have a laser sensor in front of the camera and every time something triggers the sensor it would notify the camera to snap an image. What I need help with is how do I take in the sensor input. I have a version of this in C# where I take in the PLCamera.LineStatus value and just keep checking if it is high or not I'm just not quite sure how to do this in the python version. Any and all help would be appreciated, thank you.

GuzziButz, 2018-07-18:
Hi,

As a first step you need to configure your camera for external triggering, then you need to disable pylon's default configuration handler which otherwise would configure the camera back for not using an external trigger. You simply need to add this line of code after creating the instant camera object:
`camera.RegisterConfiguration(pylon.ConfigurationEventHandler(), pylon.RegistrationMode_ReplaceAll, pylon.Cleanup_Delete)
`
`camera.RetrieveResult` would then block until image is triggered or until timeout expired.

glin123, 2018-07-18:
Hey,

Sorry, I need to clarify what I asked earlier, I want to read the status of the on board camera inputs without taking a picture. In C# I used the following lines of code:

`camera.GetCamera().Parameters[PLGigECamera.LineSelector].SetValue(PLCamera.LineSelector.Line1);
 camera.GetCamera().Parameters[PLCamera.LineInverter].SetValue(true);
 lineStatus = camera.GetCamera().Parameters[PLCamera.LineStatus].GetValue();`

 to read the input bus. Does anyone know how to do this using python?

gachiemchiep, 2018-07-19:
@glin123 

Are you trying to read camera 's Device information ? If that's what you want then you can use the following code

```
# To see which device info is available
print camera.GetDeviceInfo()
#('DeviceClass', 'DeviceFactory', 'DeviceGUID', 'DeviceIdx', 'ExposureTime', 'FriendlyName', 'FullName', 'ManufacturerInfo', 'ModelName', 'ProductId', 'SerialNumber', 'SpeedSupportBitmask', 'TransferModeKey', 'UsbDriverTypeName', 'UsbPortVersionBcd', 'VendorId', 'VendorName', 'XMLSource')]
# We can se the current ExposureTime by
print camera.ExposureTime.camera.GetValue()
```

GuzziButz, 2018-07-19:
Hi,

pypylon API is pretty straight forward and similar to the native C++ API. Just try code like this:
```
camera.LineSelector.SetValue("Line3")
e = camera.LineSelector.GetValue()
print("LineSelector Value: ", e)
camera.LineInverter.SetValue(True)
b = camera.LineStatus.GetValue()
print("Line Value: ", b)
```

closed 2018-07-19 by glin123

Thyagutamizhan, 2019-02-02:
> Hello:
>
> I'm trying to get my camera to take an image every time time a sensor is triggered. At the moment my set up is have a laser sensor in front of the camera and every time something triggers [...]

hi ,
   I'm facing the same problem,i wanna capture the image using hardware trigger and access the image using pypylon. Can i have a reference?