basler/pypylon #22: Saving color image

glin123, 2018-07-19:
Hi,

In my project I'm able to snap a picture and save to a directory of my choosing however, I would like the image to be saved as a color image. But, it keeps saving in black and white. I looked this issue up and people suggested I add this line `camera.PixelFormat.SetValue("BayerBG8")` to be able to take color images I looked at the camera manual and it does support "BayerBG8" as the color image and "Mono8" as the BnW option (camera: ac2500-20gc) but for some reason it still takes BnW images. Please help?!

gachiemchiep, 2018-07-20:
Hello sir.

Your setting value is wrong. It should be "RGB8"

At first, check your camera's available pixel format at See https://docs.baslerweb.com/#t=en%2Fpixel_format.htm

```
Choosing a Pixel Format
To choose a pixel format, set the PixelFormat parameter to one of the following values:

MonoXX (e.g., Mono10, Mono12p) (if available)
BayerXXYY (e.g., BayerBG10, BayerGR12) (if available)
YCbCr422_8, YUV422Packed, YUV422_YUYV_Packed (if available)
RGB8, BGR8 (if available)
```

Set the your camera pixel format

```
camera.PixelFormat = "RGB8"
```

gachiemchiep, 2018-07-22:
Hey @glin123 

Sorry i didn't check your camera spec.  As you can see your camera spec is as follow

[image]

And in the manual they said that the values should be set as follow: 

[image]

I used the following camera. So the "RGB8"  options work .

[image]

You did use the following option "("BayerBG8")" which does not exist in the camera specification. 
So i guess you can use the following options to get the color image

```
BayerRG8
BayerRG12
YUV
```
Hope that help.

thiesmoeller, 2018-07-24:
Hi @glin123,

BayerBG8 is a raw color format. 
PylonViewer uses the ImageFornatConverter under the hood to create a more 'classic' RGB output.

Take the [converter sample](https://github.com/basler/pypylon/blob/master/samples/utilityimageformatconverter1.py) as reference.

Use a RGB Pixelformat as output.

glin123, 2018-08-06:
Hello all, 
After I convert the image to a RGB Pixelformat how do I save the image. As of right now I am saving the image using(python) `Image.fromarray(result.Array).save(save_image_location + "tmp_img" + ".tiff")` but after I convert the image it throws an error saying "AttributeError: 'PylonImage' object has no attribute 'Array'". Is there a way I can call ImagePersistence.Save() to save the image like so(c#), `ImagePersistence.Save(ImageFileFormat.Tiff, path + "\\" + filename + ".tiff", image)`  in python?

closed 2018-08-06 by glin123