basler/pypylon #23: How to get frame exposure time?

asavpatel92, 2018-07-25:
How can I get exposure time for a particular frame after grab is finished? I can get exposure time from the camera object but if camera exposure is changed by the time grab is finished. it would be a wrong exposure time for the frame.

GuzziButz, 2018-07-26:
You can use the [data chunk ](url)Feature.

GuzziButz, 2018-07-26:
here's the correct [link ](https://github.com/basler/pypylon/issues/23)to the feature documentation.
With grabchunkimage.py you'll find a related sample in the pypylon samples collection.

asavpatel92, 2018-07-30:
@GuzziButz the link points to this same issue here. can you please point me to the documentation?

asavpatel92, 2018-07-30:
nvm. I found the documentation in the SDK.
for anyone else looking for the answer.

this is how you can access the exposure time of the frame : 
`grabResult.GetChunkNode("ChunkExposureTime").Value`

closed 2018-07-30 by asavpatel92