basler/pypylon #24: Make pyplon available on the Python Package Index (PyPI)

Niklatz, 2018-07-27:
Since you already suggest installation via pip for this package I believe it would make sense to distribute the wheels directly via the Python Package Index (https://pypi.org/). That way users would no longer be required to figure out and download the correct wheel and manually install. pip would take care of that. At the same time users could use pip to update the package easily without having to manually check the releases page on this github page.

As far as I can tell there is no package with the name pypylon on PyPI yet. So that should also be available to you.

FrankKarstens, 2018-07-27:
Hi Niklatz,
you're absolutely right, we're working on this but it will still take us a few weeks.

stefanklug, 2018-11-19:
Hi Niklas,

the problem is that pypi doesn't allow binary wheels for arm. For x86 it would be possible using manylinux, but until now we didn't investigate further.
As you correctly mentioned binary wheels are available on the github releases page.
Right now the easiest automatic install would be something like:
```
pip3 install https://github.com/basler/pypylon/releases/download/1.3.1/pypylon-1.3.1-cp35-cp35m-linux_aarch64.whl
```

NiklasKroeger, 2018-11-19:
Ah, okay. Thanks for the response!

> Right now the easiest automatic install would be something like:
> `pip3 install https://github.com/basler/pypylon/releases/download/1.3.1/pypylon-1.3.1-cp35-cp35m-linux_aarch64.whl`

Yeah that works. I just figured it would be nice I didn't have to manually check which version is the current one. Anyway, Thanks for the helpful wrapper. Works great so far for me. :+1:

matt4lieb, 2019-06-17:
> how to install pypylon for windows machine? actually it shows error messaga like this
> 
> Collecting pypylon
> Could not find a version that satisfies the requirement pypylon (from versions: )
> No matching distribution found for pypylon

That's maybe because the only published the windows whl. 
https://pypi.org/project/pypylon/#files

thiesmoeller, 2021-02-23:
Beginning with release 1.7.1 we have binary releases on all platforms, that you can install via pip.

update your pip version before by calling `python -m pip install --upgrade pip`

closed 2021-02-23 by thiesmoeller