basler/pypylon #25: Frame acquisition delay

vdmcb, 2018-07-30:
Hello, 
We are using the Basler acA2040-55um camera and we are experiencing delay of ~1s during frame acquisition using PyPylon, on Ubuntu 16.04, running this script https://github.com/basler/pypylon/blob/master/samples/grab.py
We experience the same result on both of the specified model cameras.
Using the Pylon Viewer, there is no delay.
Can you help us solve this issue?

Best regards,
Vadim Fintinari

matt-phair, 2018-07-30:
If the camera has them, have you played with the auto-exposure or auto-gain settings? If you turn off auto-gain and put the exposure down to something small, does the problem persist?

vdmcb, 2018-07-30:
In the grab.py script, from samples, the exposure and gain are not set. Are their default value set on Auto? If not, could you tell me how to set them on auto?

matt-phair, 2018-07-30:
For our camera, the values are stored on the camera itself. Easiest way to check is to open up pylon viewer and check the current properties on the camera that way. There's a little dropdown at the bottom of the list of properties where you can filter out more advanced settings. If you set it to "guru" you'll get all of them.

farzadsw, 2018-08-07:
Using the wrong grab strategy can be the cause of the problem.
Grab the images using this method  to see whether it makes any difference or not:
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

The rest of the code is in the opencv.py sample.

lcflorindo, 2020-07-27:
Hello all,

I have a similar problem but got a delay increase migrating from Ubuntu 16.04 LTS to Win10 Server 2012.
My Ubuntu runs pretty fast, close to 0.8 sec to take a picture, but on Windows it got 2.4 sec increase and misses the target (3.2 seconds total). In both systems the PylonViewer works perfectly and without any delay.

I tried also change the settings suggested above in Pylon Viewer but the camera does not keep the config, I may need do it by code. (I am using an acA1600-60gc btw)

Any ideas on how to solve this Window10 delay?

farzadsw, 2020-07-28:
I haven't used it with win10, but regarding the settings, you can set the parameters in the Pylon Viewer and save it as a file. Then in your code, you can load the setting from the file.

lcflorindo, 2020-07-29:
> I haven't used it with win10, but regarding the settings, you can set the parameters in the Pylon Viewer and save it as a file. Then in your code, you can load the setting from the file.

That sounds promissor! Is there any example code available? 
Another thing that is "fishy" for me, is about the version of the library/Pylon Viewer installed. You guys think that could have some relation with the delay? There is a "best" version to install?

thiesmoeller, 2020-07-29:
The easiest path is setting the same parameters, Just like you did in Pylon Viewer

https://docs.baslerweb.com/network-related-parameters-(gige-cameras)

barzan-hayati, 2020-11-25:
if you are using a camera that gives out raw frames (like acA2040-55uc in our case)
the delay is because you are not using "video/x-raw(memory:NVMM)".

I suggest a gstreamer pipeline like this:
`gst-launch-1.0  pylonsrc imageformat=RGB8  ! videoconvert   ! nvvideoconvert ! "video/x-raw(memory:NVMM),format=NV12"  ! m.sink_0 nvstreammux name=m batch-size=1 width=2048 height=1536 ! nvmultistreamtiler rows=1 columns=1 width=2048 height=1536 ! nvvideoconvert ! nvdsosd  ! nvegltransform  ! nveglglessink sync=false`

closed 2022-10-26 by thiesmoeller