basler/pypylon #26: Accessing user-defined values on Ace camera

campanelli-resonon, 2018-08-01:
There doesn't seem to be a way to get/set the user-defined values on an Ace camera using pypylon.

1. Am I missing something?
2. If this is indeed the case, then how hard would it be to add this functionality? (My organization might be willing to submit a PR here, but I'd like to know what we would be in for.)

NOTE: It looks like the underlying Plyon C++ SDK implements this functionality in the individual USB vs. GigE vs. ... Instant Camera classes, and I wonder why this isn't generalized, e.g.,  `GenApi::IInteger& Basler_UsbCameraParams::CUsbCameraParams_Params::UserDefinedValue`.

GuzziButz, 2018-08-02:
This should actually work. Try this code:
```
 camera.UserDefinedValueSelector.SetValue("Value1")
 camera.UserDefinedValue.SetValue(1000)
 b = camera.UserDefinedValue.GetValue()
 print("User-Defined Value: ", b)
```

campanelli-resonon, 2018-08-02:
@GuzziButz Thanks so much: That works very well. Sorry that I apparently missed this somewhere.

closed 2018-08-02 by campanelli-resonon