basler/pypylon #27: Not able to save image

cache91, 2018-08-01:
I am trying to save the image on the disk , but i am getting lot of errors. I tried to convert the array to an image and that doesnt work either. Tried running the samples and even that generates error.

I am using a linux with a basler dart

matt-phair, 2018-08-01:
Can you post the code that you've tried and the errors that you're getting?

cache91, 2018-08-02:
I tired using the code that you had with the `Image.fromarray()` in the `grab.py` example but i am getting some error for closing camera

matt-phair, 2018-08-02:
The closing error was a typo. Should be `camera.Close()` with a capital `C`. Seemed like from the other issue that you had to use `.tiff` format. Once you made those changes, do you have any other problems? The other issue is closed, so can you post the errors you're having here?

cache91, 2018-08-02:
Thanks a lot. It now works as expected :) Really appreciate your help in figuring this out !!!

closed 2018-08-02 by cache91