basler/pypylon #28: Camera parameters not accessible (Node is not readable. : AccessException thrown in node ...)

lpradeFCAP, 2018-08-06:
This may be a beginners mistake, but I have installed PyPylon using the binaries from the release page (https://github.com/Basler/pypylon/releases) and I am able to run the "grab" sample fine. However, when I try access the cameras parameter as described in the Basler C++ guide, I get an error message.

Example:
`camera.AutoGainLowerLimit.GetMin()`

Causes the message:
> Node is not readable. : AccessException thrown in node 'nA0BB' while calling 'AutoGainLowerLimit.GetMin()' (file 'IntegerT.h', line 149)

I am using a acA1600-20um (USB3.0) on Windows 10 with Spyder 3.2.8 and Pylon 5.0.12

Any help is much appreciated

GuzziButz, 2018-08-06:
Before accessing any camera feature you need to open the camera device.
`camera.StartGrabbingMax `opens the camera implicitly but if you try to access a parameter before calling StartGrabbing... you need to explicitly open the device with `camera.Open()`
Hope this helps.

lpradeFCAP, 2018-08-06:
That did fix the problem. Thanks a lot!

closed 2018-08-06 by lpradeFCAP