basler/pypylon #30: Saving converted image

glin123, 2018-08-06:
Hello all, 
I am trying to save a color image but after I convert the image from the raw "BayerBG8" format to a RGB Pixelformat it wont save. My question is after I convert the image to a RGB Pixelformat how do I save the image. As of right now I am saving the image using(python) `Image.fromarray(result.Array).save(save_image_location + "tmp_img" + ".tiff")` but after I convert the image it throws an error saying "AttributeError: 'PylonImage' object has no attribute 'Array'". Is there a way I can call ImagePersistence.Save() to save the image like so(c#), `ImagePersistence.Save(ImageFileFormat.Tiff, path + "\\" + filename + ".tiff", image)`  in python?
Camera is: ac2500-20gc

asavpatel92, 2018-08-06:
I also have exact same question. I currently use 

`img_result.GetArray(raw=True).tofile('out_raw_img')`

but I am afraid that I have lost all grab result attributes of the image for the images I have saved for quite sometime now. I now want to use image converter on those files but it doesn't work on raw files. :disappointed:

derFRUST, 2018-08-14:
This is how I managed to convert BayerBG8 to RGB and save images using Imageio (since I mostly work with NumPy arrays), maybe this helps you guys:
```
from pypylon import pylon
import imageio

# create InstantCamera and grab image
camera = pylon.InstantCamera(
    pylon.TlFactory.GetInstance().CreateFirstDevice())
grab_result = camera.GrabOne(1000)

# create NumPy array from BayerBG8 grab result
# and save Bayer raw image as grayscale BMP
image_bayer = grab_result.GetArray()
imageio.imsave('image_bayer.bmp', image_bayer)

# create and configure ImageFormatConverter
converter = pylon.ImageFormatConverter()
converter.OutputPixelFormat = pylon.PixelType_RGB8packed

# convert image to RGB, create NumPy array
# and save RGB image as color BMP
converted = converter.Convert(grab_result)
image_rgb = converted.GetArray()
imageio.imsave('image_rgb.bmp', image_rgb)
```

asavpatel92, 2018-08-15:
@derFRUST conversion is fine if you have access to `GrabResult` object. 
in my case I don't have `GrabResult` object available. because I stored raw frame on the disk using `img_result.GetArray(raw=True).tofile('out_raw_img'`.
now I want to debayer that raw frame from disk.

asavpatel92, 2018-08-16:
ok so I got it to work using `cv2`. choose `cv2` Bayer type based on your camera **pixel format.**
mine was **PixelType_BayerGR12** so I am using **COLOR_BayerGR2BGR**.

```
import numpy as np
import cv2

raw_np = np.fromfile(img_path)
raw_img = np.ndarray((IMAGE_HEIGHT, IMAGE_WIDTH), dtype=np.uint16, buffer=raw_np)
debayered_img = cv2.cvtColor(raw_img, cv2.COLOR_BayerGR2BGR)
```

Now one thing to note here is. debayered image in my case is a 16 bit image. but pixel values are 12 bit.
so image would look really dark. if you choose to convert it to 16 bit just multiply it by `2**4`

closed 2018-08-17 by glin123