basler/pypylon #31: saving images from multiple cameras

cache91, 2018-08-06:
Hi , i was able to save an image form a single camera. Can you point me in the right direction to do the same for multiple cameras ? I am using the `grabmulitplecamera.py`

asavpatel92, 2018-08-06:
use `InstantCameraArray` class for that.
here's how you can get all cameras at once.
```
from pypylon import pylon

class ConfigurationEventPrinter(pylon.ConfigurationEventHandler):
    def OnAttach(self, camera):
        print("OnAttach event")

    def OnAttached(self, camera):
        print("OnAttached event for device ", camera.GetDeviceInfo().GetModelName())

    def OnOpen(self, camera):
        print("OnOpen event for device ", camera.GetDeviceInfo().GetModelName())

    def OnOpened(self, camera):
        print("OnOpened event for device ", camera.GetDeviceInfo().GetModelName())

    def OnGrabStart(self, camera):
        print("OnGrabStart event for device ", camera.GetDeviceInfo().GetModelName())

    def OnGrabStarted(self, camera):
        print("OnGrabStarted event for device ", camera.GetDeviceInfo().GetModelName())

    def OnGrabStop(self, camera):
        print("OnGrabStop event for device ", camera.GetDeviceInfo().GetModelName())

    def OnGrabStopped(self, camera):
        print("OnGrabStopped event for device ", camera.GetDeviceInfo().GetModelName())

    def OnClose(self, camera):
        print("OnClose event for device ", camera.GetDeviceInfo().GetModelName())

    def OnClosed(self, camera):
        print("OnClosed event for device ", camera.GetDeviceInfo().GetModelName())

    def OnDestroy(self, camera):
        print("OnDestroy event for device ", camera.GetDeviceInfo().GetModelName())

    def OnDestroyed(self, camera):
        print("OnDestroyed event")

    def OnDetach(self, camera):
        print("OnDetach event for device ", camera.GetDeviceInfo().GetModelName())

    def OnDetached(self, camera):
        print("OnDetached event for device ", camera.GetDeviceInfo().GetModelName())

    def OnGrabError(self, camera, errorMessage):
        print("OnGrabError event for device ", camera.GetDeviceInfo().GetModelName())
        print("Error Message: ", errorMessage)

    def OnCameraDeviceRemoved(self, camera):
        print("OnCameraDeviceRemoved event for device ", camera.GetDeviceInfo().GetModelName())

class ImageEventPrinter(pylon.ImageEventHandler):
    def OnImagesSkipped(self, camera, countOfSkippedImages):
        print("OnImagesSkipped event for device ", camera.GetDeviceInfo().GetModelName())
        print(countOfSkippedImages, " images have been skipped.")
        print()

    def OnImageGrabbed(self, camera, grabResult):
        print("OnImageGrabbed event for device ", camera.GetDeviceInfo().GetModelName())

        # Image grabbed successfully?
        if grabResult.GrabSucceeded():
            print("SizeX: ", grabResult.GetWidth())
            print("SizeY: ", grabResult.GetHeight())
            img = grabResult.GetArray()
            print("Gray values of first row: ", img[0])
            print()
        else:
            print("Error: ", grabResult.GetErrorCode(), grabResult.GetErrorDescription())

MAX_CAMERAS = 2
tlFactory = pylon.TlFactory.GetInstance()
# Get all attached devices and raise an error if no devices found
devices = tlFactory.EnumerateDevices()
if not devices:
    raise pylon.RuntimeException("No camera present.")

# Create an array of instant cameras for the found devices and avoid
# exceeding a maximum number of devices.

cam_array = pylon.InstantCameraArray(min(len(devices), MAX_CAMERAS))
# Create and attach all Pylon Devices.
cameras = {}
images_to_grab = 10
timeout = 1000
for i, cam in enumerate(cam_array):
    cam.RegisterConfiguration(ConfigurationEventPrinter(), pylon.RegistrationMode_Append,
                                  pylon.Cleanup_Delete)
    cam.Attach(tlFactory.CreateDevice(devices[i]))
    cam.RegisterImageEventHandler(
                ImageEventPrinter(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
    camera.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)
    for i in range(images_to_grab):
        grabResult = cam.GrabOne(timeout)
```

Svato, 2019-02-18:
Hello asavpatel92
It is not working for me, did you chceck it? :D
The futex facility returned an unexpected error code.
Process finished with exit code 134 (interrupted by signal 6: SIGABRT)

Svato, 2019-02-18:
Hello asavpatel92
It is not working for me :/
The futex facility returned an unexpected error code.
Process finished with exit code 134 (interrupted by signal 6: SIGABRT)
or
Traceback (most recent call last):
  
    grabResult = cam.GrabOne(timeout)
  File "/usr/local/lib/python2.7/dist-packages/pypylon/pylon.py", line 3056, in GrabOne
('OnGrabStarted event for device ', 'acA3088-57uc')
    return _pylon.InstantCamera_GrabOne(self, *args)
_genicam.RuntimeException: Grabbing has already been started. : RuntimeException thrown (file 'InstantCameraImpl.h', line 737)

thiesmoeller, 2025-05-22:
you mixed up cam and camera in the loop...

closed 2025-05-22 by thiesmoeller