basler/pypylon #32: Camera parameters (i.e. AcquisitionStatusSelector) not available

lpradeFCAP, 2018-08-07:
When consulting the C++ programming guide from https://docs.baslerweb.com some camera parameters are used to control the "single frame acquisition" of the camera. One of the parameters used is:
camera.AcquisitionStatusSelector.SetValue(AcquisitionStatusSelector_FrameTriggerWait);

To check if a trigger has been received and a frame is available. In pypylon however, this attribute is not accessible. Is there a particular reason for this or an alternative?
Thanks,
Ludwig

FrankKarstens, 2018-08-07:
Should work. Use `camera.AcquisitionStatusSelector="FrameTriggerWait"` instead.

lpradeFCAP, 2018-08-07:
Thanks for your reply.
Maybe I didn't make the problem clear enough:
The structure "camera" does not contain an attribute named "AcquisitionStatusSelector".
Thanks,
Ludwig

GuzziButz, 2018-08-08:
I tested the above mentioned code `camera.AcquisitionStatusSelector="FrameTriggerWait"` and it worked. You must however make sure (e.g. check with pylon Viewer) that your camera actually provides this feature (not every Basler camera has an AcquisitionStatusSelector).

lpradeFCAP, 2018-08-10:
I tried the following:

`camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
camera.AcquisitionStatusSelector="FrameTriggerWait"`

and the third command causes a "Node not existing" error. Therefore, it does seem like my camera does not support that command. That is peculiar, as it is listed in the Basler guide for this camera.

Thanks for your help,
Ludwig

closed 2018-08-10 by lpradeFCAP