basler/pypylon #34: Image Conversion Error

glin123, 2018-08-17:
Hello, I'm working on a project where we are taking images and converting them to be saved. However, every once and a while the camera will throw the error: "Pylon Error: Cannot convert image. The passed source image is invalid. : InvalidArgumentException thrown (file 'ImageFormatConverter.cpp', line 77)". When this error happens the camera does not save the taken image and I'm not to sure on why this is happening and also how to fix it. I talked to Basler and they thought it might be a bandwidth issue however that is not the case. I optimized the bandwidth on the camera and it is still throwing this error. If anyone could give me some insight on this issue that would be great, thanks.

thiesmoeller, 2018-08-20:
you should call <GrabResult>.GrabSucceeded() on your retrieved/grabbed result object: 
is it True?
check this before calling the ImageConverter to check if the data you want to send to the converter is valid.

glin123, 2018-08-20:
If I add this and it did not succeed it will just skip this image and not save it correct? And also, is there a way to make an image grab successful 100% of the time, so .GrabSucceeded() will be true all the time, convert and save every image or at least really close to all the time. At this moment I would like to have it grab and save every image taken but  right now it only converts and saves around 80% of the images.

thiesmoeller, 2018-09-22:
You have to look at you system.  
Check if your converting and saving is real-time. Log timestamps during the convert/save process. 
Are you faster than frametime ?

And follow the guidelines from the manuals and camera docs. E.g. as gigevision is a nearly 100% Rx load on your system. Enlarge the receive descriptor count on the driver.
Enlarge buffer count in pylon

Make measurements and try to find the source of "trouble"

stefanklug, 2020-08-28:
I close this, as there was no activity in the last year.

closed 2020-08-28 by stefanklug