basler/pypylon #35: Configure pixel format, frame rate, bandwidth

nlbutts, 2018-08-21:
I have a C++ program that I would like to convert to pypylon. In the C++ code I do the following:
        cout << "Registering Event handlers" << endl;
        _cameras[0].RegisterImageEventHandler( new CImageEventPrinter, RegistrationMode_Append, Cleanup_Delete);
        _cameras[1].RegisterImageEventHandler( new CImageEventPrinter, RegistrationMode_Append, Cleanup_Delete);

        cout << "Setting pixel format" << endl;
        _cameras[0].PixelFormat.SetValue(Basler_UsbCameraParams::PixelFormat_BayerGR8);
        _cameras[1].PixelFormat.SetValue(Basler_UsbCameraParams::PixelFormat_BayerGR8);

        cout << "Disabling overlap mode" << endl;
        _cameras[0].OverlapMode.SetValue(Basler_UsbCameraParams::OverlapMode_Off);
        _cameras[1].OverlapMode.SetValue(Basler_UsbCameraParams::OverlapMode_Off);

        cout << "Setting frame rate" << endl;
        _cameras[0].AcquisitionFrameRate.SetValue(_desiredFPS);
        _cameras[1].AcquisitionFrameRate.SetValue(_desiredFPS);

        cout << "Setting bandwidth limit" << endl;
        _cameras[0].DeviceLinkThroughputLimitMode = Basler_UsbCameraParams::DeviceLinkThroughputLimitMode_On;
        _cameras[0].DeviceLinkThroughputLimit = _bandwidthLimit;
        _cameras[1].DeviceLinkThroughputLimitMode = Basler_UsbCameraParams::DeviceLinkThroughputLimitMode_On;
        _cameras[1].DeviceLinkThroughputLimit = _bandwidthLimit;

But I can't find a similar method of doing this in the Python code. I've tried the following:
self.cameras.PixelFormat = "BayerGR8"

This doesn't throw any errors, but I don't believe it is working correctly.

brtasavpatel, 2018-08-22:
it depends how you are initializing your camera. sharing your python code would help. I can simply do `cam.PixelFormat = "BayerRG10"` and it works.

asavpatel92, 2018-08-22:
it depends how you are initializing your camera. sharing your python code would help. I can simply do `cam.PixelFormat = "BayerRG10"` and it works.

gachiemchiep, 2018-08-23:
See https://github.com/basler/pypylon/issues/22

closed 2020-08-28 by stefanklug