basler/pypylon #36: Use pyinstaller to pack into exe file and get no device.

pineappleliubo, 2018-08-23:
# Get the transport layer factory.
tlFactory = pylon.TlFactory.GetInstance()

# Get all attached devices and exit application if no device is found.
devices = tlFactory.EnumerateDevices()

if len(devices) == 0:
    raise pylon.RuntimeException("No camera present.")

The above code runs normally.
But when pyinstaller is packaged into exe, the exception "RuntimeException No camera present." is thrown.

RoccoMatano, 2018-08-27:
Hi @pineappleliubo and @stefanklug,
i doubt that commit c34e789 fixes the reported issue. That commit fixes the problem that pypylon could not be imported at all from a pyinstaller executable. But i have to admit that i did my tests only on windows.

RoccoMatano, 2018-08-28:
Hi @pineappleliubo,
I see that you are using Python 2.7. Can you try to reproduce the issue with a more current Python version (e.g. 3.6 or 3.7)?

fncnt, 2018-10-30:
Apparently `pyinstaller` copies the pylon driver DLL files into the 'dist' directory but misses some of them.
Depending on the camera type you're using, you can manually specify which binary files `pyinstaller` should include by using the generated `.spec` file.
You can find the specific files in the [pylon Deployment Guide](https://www.baslerweb.com/en/sales-support/downloads/document-downloads/pylon-deployment-guide-software-version-5x-installation-guide/).
In my case `pyinstaller` didn't include 3 pylon files necessary for **USB3 vision** cameras which were listed in the deployment guide:
Just open `CMQI.spec` and include the missing files in the Analysis object `a`:
```
binaries=[('C:\\Program Files\\Basler\\pylon 5\\Runtime\\x64\\ProducerU3V.cti', '.'),
          ('C:\\Program Files\\Basler\\pylon 5\\Runtime\\x64\\PylonUsb_MD_VC120_V5_0_TL.dll', '.'),
          ('C:\\Program Files\\Basler\\pylon 5\\Runtime\\x64\\uxapi_v10.dll', '.')],
```
You probably need some other files if you're not using a **USB3 vision** camera.
Then you can freeze your script by running:
```
pyinstaller CMQI.spec
```

RoccoMatano, 2018-10-30:
I think @fncnt pointed in the right direction. But i would solve it this way:
```
# Pyinstallers dependency analysis will not detect that pylons TL DLLs are
# required (since they are searched and loaded from machine code at runtime).
# Consequently it would not include them in the archive. This is amended by
# simply adding all DLLs from the pypylon directory to the list of binaries.

import pypylon
import pathlib
pypylon_dir = pathlib.Path(pypylon.__file__).parent
pylon_dlls = [(str(dll), '.') for dll in pypylon_dir.glob('*.dll')]

a = Analysis(
    ...
    binaries=pylon_dlls,        # make sure pylons TL DLLs are added
    ...
    )
```

ByteArts, 2019-02-19:
I am having this same problem. I've tried the solutions suggested by @fncnt and @RoccoMatano , but when running the executable generated by pyinstaller, no camera is ever found. The same code works fine when run with Python and finds my camera.

Details:
Using Python 2.7.15 (64 bits), Windows 8.1. Camera is a daA1600-60um. I have the most recent version of pypylon (V1.3.1) installed. 

I've modified the .spec file to include ALL the files in the Runtime/x64 folder. The program file I'm using is this (filename is enum_cameras.py):

    from pypylon import pylon

    tlFactory = pylon.TlFactory.GetInstance()

    # Get all attached devices and exit application if no device is found.
    devices = tlFactory.EnumerateDevices()

    if len(devices) == 0:
        raise pylon.RuntimeException("No camera present.")
    else:
        print '%d camera(s) found' % len(devices)

When I run the executable, I get this output:
**Traceback (most recent call last):
  File "enum_cameras.py", line 9, in <module>
_genicam.RuntimeException: No camera present.
[4084] Failed to execute script enum_cameras**

Is there a way to get more detailed debugging information?

yaochuang001, 2019-05-13:
I have encountered the same problem, what should I do according to your method?@RoccoMatano

ByteArts, 2019-05-13:
Here's the .spec file I used - maybe this will help. You have pass the .spec file to pyinstaller instead of the .py file, so in this case I use "pyinstaller example.spec"

    # -*- mode: python -*-
    
    block_cipher = None
    
    # Pyinstallers dependency analysis will not detect that pylons TL DLLs are
    # required (since they are searched and loaded from machine code at runtime).
    # Consequently it would not include them in the archive. This is amended by
    # simply adding all DLLs from the pypylon directory to the list of binaries.
    
    import pypylon
    import pathlib
    pypylon_dir = pathlib.Path(pypylon.__file__).parent
    pylon_dlls = [(str(dll), '.') for dll in pypylon_dir.glob('*.dll')]
    
    
    a = Analysis(['qcam_main.py'], 
        pathex=['E:\\Projects_Quantel\\QCam\\software\\quantelcam'],
        binaries=pylon_dlls,
        datas=[('control_wnd_main.ui', '.')],
        hiddenimports=[],
        hookspath=[],
        runtime_hooks=[],
        excludes=[],
        win_no_prefer_redirects=False,
        win_private_assemblies=False,
        cipher=block_cipher,
        noarchive=False)
    pyz = PYZ(a.pure, a.zipped_data,
         cipher=block_cipher)
    exe = EXE(pyz,
         a.scripts,
         [],
         exclude_binaries=True,
         name='qcam_main',
         debug=False,
         bootloader_ignore_signals=False,
         strip=False,
         upx=True,
         console=True )
    coll = COLLECT(exe,
           a.binaries,
           a.zipfiles,
           a.datas,
           strip=False,
           upx=True,
           name='qcam_main')

yaochuang001, 2019-05-14:
> Here's the .spec file I used - maybe this will help. You have pass the .spec file to pyinstent

> Here's the .spec file I used - maybe this will help. You have pass the .spec file to pyinstaller instead of the .py file, so in this case I use "pyinstaller example.spec"
> 
> ```
> # -*- mode: python -*-
> 
> block_cipher = None
> 
> # Pyinstallers dependency analysis will not detect that pylons TL DLLs are
> # required (since they are searched and loaded from machine code at runtime).
> # Consequently it would not include them in the archive. This is amended by
> # simply adding all DLLs from the pypylon directory to the list of binaries.
> 
> import pypylon
> import pathlib
> pypylon_dir = pathlib.Path(pypylon.__file__).parent
> pylon_dlls = [(str(dll), '.') for dll in pypylon_dir.glob('*.dll')]
> 
> 
> a = Analysis(['example.py'], 
>     pathex=['E:\\Projects\\example'],
>     binaries=pylon_dlls,
>     datas=[],
>     hiddenimports=[],
>     hookspath=[],
>     runtime_hooks=[],
>     excludes=[],
>     win_no_prefer_redirects=False,
>     win_private_assemblies=False,
>     cipher=block_cipher,
>     noarchive=False)
> pyz = PYZ(a.pure, a.zipped_data,
>      cipher=block_cipher)
> exe = EXE(pyz,
>      a.scripts,
>      [],
>      exclude_binaries=True,
>      name='example',
>      debug=False,
>      bootloader_ignore_signals=False,
>      strip=False,
>      upx=True,
>      console=True )
> coll = COLLECT(exe,
>        a.binaries,
>        a.zipfiles,
>        a.datas,
>        strip=False,
>        upx=True,
>        name='example')
> ```

I am not familiar enough with pyinstaller, and I am too careless. Now it works.thank you very much

lee2430, 2019-06-27:
@RoccoMatano Thank you very much. I encountered the same problem, and directly solve it with your method. You saved me much time. Thanks again!

binglebai, 2020-01-19:
I added it to the .spec and can print() all .dll
but But it's still wrong (NO Device)
I dont why 
[('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\GCBase_MD_VC141_v3_1_Basler_pylon.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\GenApi_MD_VC141_v3_1_Basler_pylon.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\gxapi_v11.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\log4cpp_MD_VC141_v3_1_Basler_pylon.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\Log_MD_VC141_v3_1_Basler_pylon.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\MathParser_MD_VC141_v3_1_Basler_pylon.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\msvcp140.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\NodeMapData_MD_VC141_v3_1_Basler_pylon.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\PylonBase_v5_2.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\PylonCamEmu_v5_2_TL.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\PylonGigE_v5_2_TL.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\PylonGUI_v5_2.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\PylonUsb_v5_2_TL.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\PylonUtility_v5_2.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\uxapi_v11.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\vcruntime140.dll', '.'), ('C:\\Users\\amp\\PycharmProjects\\untitled1\\venv\\Lib\\site-packages\\pypylon\\XmlParser_MD_VC141_v3_1_Basler_pylon.dll', '.')]

binglebai, 2020-01-19:
I added it to the .spec and can print() all .dll
but But it's still wrong (NO Device)
I dont know why
[('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\GCBase_MD_VC141_v3_1_Basler_pylon.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\GenApi_MD_VC141_v3_1_Basler_pylon.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\gxapi_v11.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\log4cpp_MD_VC141_v3_1_Basler_pylon.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\Log_MD_VC141_v3_1_Basler_pylon.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\MathParser_MD_VC141_v3_1_Basler_pylon.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\msvcp140.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\NodeMapData_MD_VC141_v3_1_Basler_pylon.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\PylonBase_v5_2.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\PylonCamEmu_v5_2_TL.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\PylonGigE_v5_2_TL.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\PylonGUI_v5_2.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\PylonUsb_v5_2_TL.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\PylonUtility_v5_2.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\uxapi_v11.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\vcruntime140.dll', '.'), ('C:\Users\amp\PycharmProjects\untitled1\venv\Lib\site-packages\pypylon\XmlParser_MD_VC141_v3_1_Basler_pylon.dll', '.')]

pwuertz, 2020-08-20:
Just hit this problem as well. I tried fixing the problem by creating `hook-pypylon.py` and using the PyInstaller utility methods to do the work. `collect_dynamic_libs` indeed finds all the dll/so files and places them in the `pypylon` subfolder, just like the original layout. Unfortunately, for some reason the installed process requires all the pypylon libs to be installed into the deploy root folder (i.e. not in ./pypylon). This hook script does the trick for me:

```python
# hook-pypylon.py
from PyInstaller.utils.hooks import collect_dynamic_libs

binaries = [
    (lib, ".") for lib, path in
    collect_dynamic_libs("pypylon")
]
```

pwuertz, 2021-02-01:
I'm not sure if this is due to a specific change in `pypylon` or `PyInstaller`, but I had to modify the `hook-pypylon.py` to get a working bundle with current versions:
```python
# hook-pypylon.py (requires python 3.8)
import importlib.metadata

pypylon_dist = importlib.metadata.distribution("pypylon")
base_path = pypylon_dist.locate_file(".")
dynamic_libs = [
    (base_path / f) for f in pypylon_dist.files
    if f.suffix.lower() in (".so", ".dll", ".dylib")
]

binaries = [(lib, ".") for lib in dynamic_libs]
```