basler/pypylon #37: Using the software trigger on 4 different acA1920-40gc

LucasVandroux, 2018-08-27:
Hello everybody,

I have 4 different acA1920-40gc cameras on the same PoE switch and I would like to trigger the cameras at the same time. 

I first used an `InstantCameraArray` and then going through it doing a `GrabOne(100)` on each of the cameras but this is not optimal and there is some delay.

What would be the best to do it? I saw that there is function called `ExecuteSoftwareTrigger()` but the example is only for a single camera case. How should I go through the `InstantCameraArray` and make sure all the cameras are taking a picture at the same time?

Thank you very much for your help,

RoccoMatano, 2018-08-28:
I think your options are to use one of these:

*[Synchronous Free Run](https://docs.baslerweb.com/#t=en%2Fsynchronous_free_run.htm)
*[Action Commands](https://docs.baslerweb.com/#t=en%2Faction_commands.htm)
*[Scheduled Action Commands](https://docs.baslerweb.com/#t=en%2Fscheduled_action_commands.htm)

LucasVandroux, 2018-08-28:
@RoccoMatano, thanks a lot for your fast feedback. This seems indeed to go in the right direction.

What about the 'ExecuteSoftwareTrigger()' why can't this function be used for this purpose?

RoccoMatano, 2018-08-29:
Since `ExecuteSoftwareTrigger()` is a method of `InstantCamera`, it only deals with a single camera. There is no `ExecuteSoftwareTrigger()` method in the class `InstantCameraArray` (because that cannot be implemented in a generic way for all camera models),

LucasVandroux, 2018-08-30:
@RoccoMatano Thank you, for the precision. It is a lot clearer now.

stefanklug, 2020-08-28:
Looks solved...

closed 2020-08-28 by stefanklug