basler/pypylon #41: Simultaneous capturing on two 1920-25uc cameras with software signals

chtlp, 2018-09-30:
Hi we have two acA1920-25uc USB cameras and we want to take a sequence of images simultaneously with the two cameras for stereo vision.  The camera model should support the software trigger, signal pulse, trigger source and other related features according to the documentation:
https://docs.baslerweb.com/#t=en%2Ftrigger_software.htm
https://docs.baslerweb.com/#t=en%2Fsoftware_signal_pulse.htm
https://docs.baslerweb.com/#t=en%2Ftrigger_source.htm

According to the sample code, we can call software trigger sequentially on the two cameras, as in the following code:
https://github.com/basler/pypylon/blob/master/samples/grabusinggrabloopthread.py

```
# Example of an image event handler.
class SampleImageEventHandler(pylon.ImageEventHandler):
    def __init__(self):
        super().__init__()
        self.grab_times = [0.0]

    def OnImageGrabbed(self, camera, grabResult):
        self.grab_times.append(time.time())
        if len(self.grab_times) % 50 == 0:
            logging.info("camera %s, CSampleImageEventHandler::OnImageGrabbed called.",
                         camera.DeviceInfo.GetSerialNumber())

def test_pylon_multiple_grab_loop_thread(num_cameras=None):
    num_cameras = parse_args(num_cameras)
    cameras = []
    handlers = []
    device_info_list = pylon.TlFactory.GetInstance().EnumerateDevices()
    for info in device_info_list[:num_cameras]:
        cam = pylon.InstantCamera(
            pylon.TlFactory.GetInstance().CreateFirstDevice(info))
        handlers.append(SampleImageEventHandler())
        cam.RegisterConfiguration(pylon.SoftwareTriggerConfiguration(),
                                  pylon.RegistrationMode_ReplaceAll,
                                  pylon.Cleanup_Delete)
        cam.RegisterConfiguration(ConfigurationEventPrinter(),
                                  pylon.RegistrationMode_Append,
                                  pylon.Cleanup_Delete)
        cam.RegisterImageEventHandler(handlers[-1],
                                      pylon.RegistrationMode_Append,
                                      pylon.Cleanup_Delete)
        cam.StartGrabbing(pylon.GrabStrategy_OneByOne,
                          pylon.GrabLoop_ProvidedByInstantCamera)
        cameras.append(cam)

    ind = 0
    while True:
        ind += 1
        t0 = time.time()
        for cam in cameras:
            # Execute the software trigger.
            # Wait up to 100 ms for the camera to be ready for trigger.
            if cam.WaitForFrameTriggerReady(
                    100, pylon.TimeoutHandling_ThrowException):
                pass
        for h in handlers:
            # all cameras should have been triggered the same number of times
            assert len(h.grab_times) == len(handlers[0].grab_times)
        last_grab_times = [h.grab_times[-1] for h in handlers]
        t1 = time.time()
        for cam in cameras:
            cam.ExecuteSoftwareTrigger()
        t2 = time.time()
        if ind % 20 == 0:
            logging.info("waiting for cameras took %.2f ms, trigger takes %.2f ms, "
                         "total %.2f ms, last_grab_times_span %.2f ms",
                         (t1-t0) * 1000, (t2-t1) * 1000, (t2-t0) * 1000,
                         (max(last_grab_times) - min(last_grab_times)) * 1000)
```

However, we find that each trigger takes 0.5ms, so the cameras won't capture images at exactly the same time. So our questions are:
1. Can we take sequence of images simultaneously on two acA1920-25uc (or other Basler USB3 cameras) through the same software trigger, signal pulse, etc?
2. We understand that software signals might cause some latencies, so what's the expected time delay between the two cameras using this approach?

thiesmoeller, 2018-10-04:
## 1)
On USB there is no generic way to broadcast a software trigger to multiple devices.[ On GigEVision cameras you could use the Action-Command feature]

To take images simultaneously between two USB3 cameras in a generic way, you have to wire them up on their I/O Ports.
Escpecially as you are talking about a stereo camera use case, I assume, that they are physically close to each other.

If you want to use software-triggered acqisition, you could:

 ### camera A 
toggle the one output line of this camera by software [LineSource UserOutput ]
     the output line is wired to an input line on the same camera    
     _and_
      and to an input line on the second camera B
      TriggerSource is the InputLine

 ### camera B

TriggerSource is the InputLine

By working this way, the two camera will start the acquisition in the exact same moment.

(how to use the TTL/GPIO ports, look up the electrical requirements in the camera manual )

## 2)
like explained above, there is no generic way to broadcast on USB. So depending on your PC, your OS and other factors from the environment, the delay between the two "writes" to the cameras could have some latency and definitly some jitter.
To get precisision you should use the wired solution.

thiesmoeller, 2020-11-30:
@chtlp  no further action since long time ... will close the issue

closed 2020-11-30 by thiesmoeller