basler/pypylon #42: Changing the Pixel Format

gust2363, 2018-10-01:
Hello,
I've got 4 daA2500-14um cameras. I already managed to set auto exposure and gain to "Off". Now I want to change the pixel format from default "Mono8" to "Mono12". The identifier for turning auto exposure of is a simple "Off", but the pixel format can't be turned to "Mono12" by cam.PixelFormat.SetValue("Mono12").
Can somebody tell me what the correct identifier for this action is?

Here is my code:
````
# imports
import os
os.environ["PYLON_CAMEMU"] = "3"
from pypylon import genicam
from pypylon import pylon
import sys
import numpy as np
import scipy.misc
import cv2

# preface
exitCode = 0
maxCamerasToUse = 4
countOfImagesToGrab = 1
try:

    # get the transport layer factory
    tlFactory = pylon.TlFactory.GetInstance()

    devices = tlFactory.EnumerateDevices()
    if len(devices) == 0:
        raise pylon.RUNTIME_EXCEPTION("No camera present")

    cameras = pylon.InstantCameraArray(min(len(devices), maxCamerasToUse))

    for ii, cam in enumerate(cameras):
        cam.Attach(tlFactory.CreateDevice(devices[ii]))

        print(cam.GetDeviceInfo().GetModelName(), "-", cam.GetDeviceInfo().GetSerialNumber())

        # start grabbing
        cam.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
        
        
        # set parameters
        # pixel format
        print(cam.PixelFormat.GetValue())
        cam.PixelFormat.SetValue("Mono 12")
        print(cam.PixelFormat.GetValue())
        # exposure time
        cam.ExposureAuto.SetValue("Off")
        cam.ExposureTime.SetValue(100.0)
        # gain
        cam.GainAuto.SetValue("Off")
        cam.Gain.SetValue(1.0)
        
        cam.Gain.SetValue(float(11))
        print(cam.Gain.GetValue())
        print(cam.ExposureTime.GetValue())

        grabResult = cam.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
        cameraContextValue = grabResult.GetCameraContext()
        print("Grab succeeded: ", grabResult.GrabSucceeded())

        if grabResult.GrabSucceeded() == True:
            img = grabResult.GetArray()
            path = "D:\Python\PyBaslerMultiCam\PyBaslerMultiCam"
            filename = cam.GetDeviceInfo().GetSerialNumber()
            filetyp = "png"
            fullpath = os.path.join(path, filename + "." + filetyp)
            scipy.misc.imsave(fullpath, img)

        cam.StopGrabbing()

except genicam.GenericException as e:
    # error handling
    print("An exception occurred.", e.GetDescription())
    exitCode = 1

sys.exit(exitCode)
````

GuzziButz, 2018-10-01:
Hi,

you almost did it correctly. it must be
`cam.PixelFormat.SetValue("Mono12")`
and not
`cam.PixelFormat.SetValue("Mono 12")`

gust2363, 2018-10-02:
Hi,
already tried that, doesn't work either. Gives an exception: 
  Nachricht = AttributeError("'AccessException' object has no attribute 'GetDescription'",)
  Quelle = d:\python\pybaslermulticam\pybaslermulticam\pybaslermulticam\module1.py
  Stapelüberwachung:
  File "d:\python\pybaslermulticam\pybaslermulticam\pybaslermulticam\module1.py", line 92, in <module>
    print("An exception occurred.", e.GetDescription())

thiesmoeller, 2018-10-05:
Hi,

to see what pixelformats a device supports you can use:

`print(cam.PixelFormat.Symbolics)`

In your programming sample, I'm pretty sure, that you are not talking to your dart cameras, but to the pylon internal device emulator. Not sure if this device supports Mono12

please remove the line
`os.environ["PYLON_CAMEMU"] = "3"`

thiesmoeller, 2018-10-08:
Hi,

pylon has a camera emulator integrated. 
you can enable it by setting the environment variable

`PYLON_CAMEMU` to a value >="1".

As you don't need the emulator I asked, why you have it enabled.
But this was not the error in your example... 
It is not that obvious but simple to solve:

a) you have to `Open()` the devices before use. Otherwise you are only allowed to read, but not to write parameters.

```
        ...
        cam.Attach(tlFactory.CreateDevice(devices[ii]))
        cam.Open()
        ...
```

b) you can't change the PixelFormat after you called `StartGrabbing(xx)`. This has the following reason:
* startgrabbing will setup all DMA and memory ressources to grab the image.
* if you change the PixelFormat you could potentially modify the memory size of one image. Mono8->Mono12 doubles the memory footprint.

so just move the

```
  # start grabbing
  cam.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

```

after setting all image parameters. Setting Gain and Exposuretime is not touching the image size, so they worked.

If you use the PylonViewer you would have seen, that the PixelFormat control is blocked after you start image acquisition.

gust2363, 2018-10-09:
Thank you, works fine!

closed 2018-10-09 by gust2363