basler/pypylon #43: Format converter not working

lievenleroy, 2018-10-02:
I'm trying to deBayer images received from a  Basler camera. It seems pylon.ImageFormatConverter() should be able to do this.

However, even the sample in utilityimageformatconverter.py doesn't work for me.

Initially when executed it generates a code rot error:
AttributeError: 'PylonImage' object has no attribute 'Array'

Replacing the script's reference to image.Array with image.GetArray() resolves this, but the resulting output array is all 0s.

Any suggestions? Thanks!

thiesmoeller, 2018-10-04:
Please use the [utilityimageformatconverter1.py](https://github.com/basler/pypylon/blob/master/samples/utilityimageformatconverter1.py) example as reference.  

The sample you chose ([utilityimageformatconverter.py](https://github.com/basler/pypylon/blob/master/samples/utilityimageformatconverter.py)) is using the PylonImage as an extra step, which is creating the trouble you have.

lievenleroy, 2018-10-05:
Hello thiesmoeller. Thank you for your suggestion! That script has a number of issues as well, but the relevant piece did get me up and running.

closed 2018-10-05 by lievenleroy