basler/pypylon #44: Problem with image.GetArray() after using converter

LuisPatino, 2018-10-03:
I am looking to convert a pylon image into a compatible OpenCV image. The example script gives the lines of code:

image = converter.Convert(grabResult)
img = image.GetArray()

However I get the following error:
AttributeError: 'PylonImage' object has no attribute 'GetArray'

The workaround we implemented is manually convert to a numpy array using image.GetBuffer():
image = converter.Convert(grabResult)
img = numpy.frombuffer(image.GetBuffer(), dtype=numpy.uint8)
img = numpy.reshape(img,(image.GetHeight(), image.GetWidth(), 3))
This works very well.

thiesmoeller, 2018-10-04:
Hi Luis,
please check with current versions. The version you work with is way too old.
As we are now on pypi you can easily use:
`pip.exe install pypylon --upgrade`
to stay up to date.

thiesmoeller, 2018-10-11:
Hi,

You see only the windows builds on pypi, because only manylinux1 wheels are allowed on pypi. We use another build platform on travis for the linux builds.
Please use the release page on GitHub:

[Release 1.3.1](https://github.com/basler/pypylon/releases/tag/1.3.1)

closed 2020-08-28 by stefanklug