basler/pypylon #45: Unable to grab images when accessing a remote camera

srinivasnks, 2018-10-04:
I am setting the 'IpAddress' property in device info as follows:
```
    ptl = factory.CreateTl('BaslerGigE')
    empty_camera_info = ptl.CreateDeviceInfo()
    empty_camera_info.SetPropertyValue('IpAddress', ip)
    camera_device = factory.CreateDevice(empty_camera_info)
    camera = pylon.InstantCamera(camera_device)
    # Print the model name of the camera.
    print("Using device ", camera.GetDeviceInfo().GetModelName())
```
I notice that if I am inside the same subnet as the camera, I am able to connect to the camera and grab images. But if I am outside the subnet of the camera, I am unable to grab images. I am still able to discover the camera though.

# python3 grab_ip.py
Using device  acA2440-20gc
An exception occurred.
Traceback (most recent call last):
  File "grab_ip.py", line 50, in <module>
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
  File "/usr/local/lib/python3.5/dist-packages/pypylon/pylon.py", line 2951, in RetrieveResult
    return _pylon.InstantCamera_RetrieveResult(self, *args)
_genicam.TimeoutException: Grab timed out. The acquisition is not started. : TimeoutException thrown (file 'InstantCameraImpl.h', line 1048)

Do I need any other setting? I can provide tcpdumps of the communication between the camera and my pc.

srinivasnks, 2018-10-04:
The zip file contains 2 files - host.dump and vpn.dump.
Host.dump is when I am in the same subnet as the camera and image grab is successful.
vpn.dump is when I on a different network and i VPN into the network of the camera.
[Archive.zip](https://github.com/basler/pypylon/files/2444760/Archive.zip)

thiesmoeller, 2018-10-08:
Hi @srinivasnks,
in the vpn.dump is not a single GVCP communication with the camera. Is it recorded on the application side of the vpn? Or on the other side of the vpn?

do you have wireshark available on your systems? wireshark as full decoding for GEV integrated

r-f-g, 2019-03-05:
Same for me. I try set destination IP with `camera.GevSCDA`, but any Grab function will set default ip address.

matt4lieb, 2019-06-27:
@r-f-g , do you have an example? Because camera.GevSCDA = "192.168.3.2" is no working. 
`*** NotImplementedError: Wrong number or type of arguments for overloaded function on 'IInteger_SetValue`

lcflorindo, 2019-12-19:
I was able to connect with the camera but it did not grabbed the image using this format:
```
from pypylon import pylon
from pypylon import genicam
import sys

countOfImagesToGrab = 1

try:
    ip_address = '198.162.0.17'
    factory = pylon.TlFactory.GetInstance()
    ptl = factory.CreateTl('BaslerGigE')
    empty_camera_info = ptl.CreateDeviceInfo()
    empty_camera_info.SetPropertyValue('IpAddress', ip_address)
    camera_device = factory.CreateDevice(empty_camera_info)
    camera = pylon.InstantCamera(camera_device)
    
    #camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice(info))
    camera.Open()
```

Then I got the grab time out as a result:
```
Using device  acA1600-60gc
An exception occurred.
---------------------------------------------------------------------------
TimeoutException                          Traceback (most recent call last)
<ipython-input-2-0445b748435d> in <module>
     30         # Wait for an image and then retrieve it. A timeout of 5000 ms is used.
---> 31         grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
     32 

~/anaconda3/envs/pycaml8/lib/python3.6/site-packages/pypylon/pylon.py in RetrieveResult(self, *args)
   3587         """
-> 3588         return _pylon.InstantCamera_RetrieveResult(self, *args)
   3589 

TimeoutException: Grab timed out. The acquisition is not started. : TimeoutException thrown (file 'InstantCameraImpl.h', line 1031)
```

Is there any relation with Packet Size or Packet Delay values?

closed 2024-01-31 by SMA2016a