basler/pypylon #47: Support for IEEE1394 camera

michuu121, 2018-10-09:
Should pypylon work with a IEEE 1394 "FireWire" camera?
I have two basler cameras, one for GigE (acA2500-14gm) and one for IEEE1394 (scA1400-17fm). Both work in pylon viewer (5.1.0.12681), but with pypylon I can only detect the GigE camera using tlFactory.EnumerateDevices() (in which only one camera gets enumerated) or TlFactory.GetInstance().CreateFirstDevice(). If I only connect the IEEE1394 camera it is not detected by pypylon as well.
I have pypylon-1.3.1 with Python 2.7.13 and 64-bit Windows 7.

BRougeHub, 2018-10-26:
I'm facing the same problem. Has anyone got a solution?

michuu121, 2018-10-29:
@BRougeHub, I have _not_ found the solution to the problem. For now, I am using a replacement solution which involves using the pynivision package (https://pypi.org/project/pynivision/). The package is just a wrapper around the National Instruments IMAQ libraries and others, thus it allows one to use the basler cameras just like in LabVIEW (although it seems to have some deficiencies, for example the GigE camera does not work...). I hope to use the firewire camera with (py)pylon though, perhaps it could run at a slightly higher frame rate.

BRougeHub, 2018-10-31:
Okay. Seems like you need to have the NI Vision Acquisition license to get the library to work. I don't have that license. 
But thanks for your help. I appreciate it.

thiesmoeller, 2019-02-01:
as we currently have no plans to support IEEE1394 for ieee1394, here is an untested idea to solve the issue:

if you build pypylon on your own, you can extend the code in the setup.py that copies the transport layer libs from the PylonSDK. The 1394 libs are just not copied at the moment
It should even be possible to copy the required 1394 TL libraries from a full basler PylonSDK [ same version as the one pypylon was build with] installation into the python/lib/site-packages/pypylon folder

The pypylon is transportlayer agnostic. So as long you don't need any TL specific functionality, it should work.

thiesmoeller, 2020-11-30:
@michuu121 , @BRougeHub : as we still have no official IEEE1394 support in and my proposal got no feedback from you, we'll close the issue

closed 2020-11-30 by thiesmoeller