basler/pypylon #48: Exception handling

pfilz0, 2018-10-11:
The attribute 'GetDescription' is not present for all exceptions handled by genicam. E. g. when trying to grab an image while the camera is controlled by another application, a RuntimeException is thrown. If handled as in the samples via:
```python
except genicam.GenericException as e: 
    print("An exception occurred. ", e.GetDescription())`
```
, python throws an AttributeException

campanelli-resonon, 2018-10-11:
I think I've hit this too and have resorted to using `str(e)` in general.

pfilz0, 2018-10-11:
True, that would work. But it would still be nice if the args attribute were copied to GetDescription()

thiesmoeller, 2020-11-30:
@pfilz0 @campanelli-resonon : will close this is issue as it seems to be solved. For any enhancement you made, please send a pull request with the changes

closed 2020-11-30 by thiesmoeller