basler/pypylon #49: Bandwidth errors on Raspberry Pi

jveitchmichaelis, 2018-10-16:
I'm running PyPylon on a Pi 3B+. I can capture images using the basic `grab.py` example, however if I plug another camera in and acquire, I'm no longer able to capture images. For example if I run this script:

```
import cv2
import time

cap = cv2.VideoCapture(0 + cv2.CAP_V4L2)

_, _ = cap.read()
time.sleep(10)
```

Even after the other camera has finished capturing (i.e. I wait for the delay), I get:

```
pi@raspberrypi:~/pypylon/samples $ python3 grab.py
Using device  daA1600-60uc
Error:  3791651083 The image stream is out of sync.
Error:  3791651083 The image stream is out of sync.
Error:  3791651083 The image stream is out of sync.
Error:  3791651083 The image stream is out of sync.
Error:  3791651083 The image stream is out of sync.
Error:  3791651083 The image stream is out of sync.
Error:  3791651083 The image stream is out of sync.
Error:  3791651083 The image stream is out of sync.
Error:  3791651083 The image stream is out of sync.
Error:  3791651083 The image stream is out of sync.
```

It's definitely not the same camera, because the Dart doesn't enumerate as a V4L device. Is there some configuration I can change?

The script works perfectly well when run on its own.

jveitchmichaelis, 2018-10-16:
> Assuming *the other camera* is an UVC camera, which is using ISO
> transmission on USB.
>
> ISO has prio on BULK used by the usb3vision protocol.
>
> The errors you see are the final effect of a camera not being able to
> deliver the video data to the Raspberry Pi
>
> Steps:
> A) Reduce the ISO bandwidth of the other camera.
> B) Adust Devicelinkthroughputlimit to the remaining bandwidth ( 480Mbit -
> ISO bandwidth )

Hi Thies,

Yep it's a UVC camera. I assume that opencv must be reading in a stream
continuously, even if you don't call "read". Though it's only a 16-bit VGA
camera so it shouldn't be saturating the bandwidth. I don't think there's
an obvious way of bandwidth limiting in opencv. It's using the V4L backend
so perhaps there's a better capture library to use.

Depending on the latency involved with releasing the uvc camera, I could
probably alternate capture between the two.

Thanks!

jveitchmichaelis, 2018-10-17:
Thanks - that makes a bit more sense. The UVC camera is pulling 60fps, which is around 320mbps. I'll see if I can halve that and run the Dart with what's left over.

Thyagutamizhan, 2019-03-16:
hi bro,
   lemme know how did you installe the pypylon package in raspberry pi 3b+?

jveitchmichaelis, 2019-03-16:
@Thyagutamizhan just use the arm 32 bit hard float package from here (either the deb file or the tarball).

https://www.baslerweb.com/en/sales-support/downloads/software-downloads/#type=pylonsoftware;version=all;os=linuxarm

The deb installer is easiest "sudo dpkg -I pylon*.deb"

charlie-plenty, 2019-05-08:
@Thyagutamizhan Did you manage to get everything up and running okay from within a Docker container?

jveitchmichaelis, 2019-11-02:
Late reply, but yes I did in the end. It works quite well, I just limited the Dart bandwidth to 10MB/s which let me capture at full speed from the other camera and also save to an external USB stick.

closed 2019-11-02 by jveitchmichaelis