basler/pypylon #52: Cannot import pypylon.pylon on Windows when running as system account

ronaldsipkema, 2018-10-26:
As part of a CI workflow we had tests trying to import pypylon.pylon, which crashed the interpreter. I had no success trying to recreate the crash until I ran my command line as the system user (using sysinternals psexec: psexec -i -s cmd.exe); our CI agent also runs as the system user. Simply starting the python interpreter and calling 'import pypylon.pylon' crashes the interpreter process. Attaching a debugger showed an uncaught exception:

Unhandled exception at 0x00007FFE2D7AA388 in python.exe: Microsoft C++ exception: GenICam_3_0_Basler_pylon_v5_0::RuntimeException at memory location 0x00000000009DDEA0.

The last (non c runtime or kernel) call on the stack was to Log_MD_VC120_v3_0_Basler_pylon_v5_0.dll!00007ffe0cf6261d()

Anyway, not sure if this is a pypylon thing or Pylon in general, and also not sure if this is a major issue, but it seems things do not work as expected when run using a system user account. Note that there was no Pylon software installed on the system where this crashed (it's a test server).

thiesmoeller, 2020-11-30:
@ronaldsipkema is this still an issue? We've reproduces the behaviour .... but found no root cause.

ronaldsipkema, 2020-12-16:
> @ronaldsipkema is this still an issue? We've reproduces the behaviour .... but found no root cause.

I'm not really affected by this issue at the moment; I usually do not run as a system user. Regarding the cause: not really possible for me to determine (without debugging symbols), but my guess would be that it has something to do with file system access rights and logging (since the last call on the stack was from Log_MD_VC120_v3_Basler-pylon_v5_0.dll.

thiesmoeller, 2025-05-22:
Closing this issue as the original reporter is no longer affected and does not run as a system user. There is no active user impact, and the root cause could not be reproduced or confirmed. If anyone encounters this problem in the future, please open a new issue with updated details. Thank you for your feedback and input!

closed 2025-05-22 by thiesmoeller