basler/pypylon #53: The acA3800-14uc can't adjust 1us per unit

HughLeoWu, 2018-10-29:
Hi Sir

As title, I use the acA3800-14uc camera, and then I run unittest "./pypylon/tests/pylon_tests/usn/calltest.py", but it failed.

The failed message was:
F....
======================================================================
FAIL: test_exposure_time (__main__.CallTestSuite)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\Users\linka\Desktop\dip_aoi\pypylon_github\pypylon\tests\pylon_tests\usb\calltest.py", line 58, in test_exposure_time
    self.assertEqual(cam.ExposureTime.Max, cam.ExposureTime.Value)
AssertionError: 1600000.0 != 1599990.0

and then I hard code to change "cam.ExposureTime.Value+10.0", it still failed.

The failed message was:
Traceback (most recent call last):
  File "C:\Users\linka\Desktop\dip_aoi\pypylon_github\pypylon\tests\pylon_tests\usb\calltest.py", line 61, in test_exposure_time
    self.assertEqual(1000, cam.ExposureTime.Value)
AssertionError: 1000 != 1015.0

I opened "pylon 5.1.0 Camera Software Suite Windows" to adjust camera exposure time.
The exposure time seems 35us unit of the time.

I checked the document "https://graftek.biz/system/files/2576/original/Basler_Ace_USB_3.0_Manual.pdf?1479057814"
The acA3800-14uc is 1us of the time, and the maximum value is 1600000

Were they inconsistent? or I install wrong windows driver?

thiesmoeller, 2018-10-29:
This is a mistake in the test itself. It take the wrong assumption, that 1000µs can be set on every device.
[ test has to be fixed accordingly ]

Some devices have an increment for the exposuretime != 1µs. The reason for getting the error in the assert is:
1000µs is not a multiple of 35µ. In Genicam all Floating-Point features will round to the nearest internal integer representation. To get the actual value applied you have to first set, then read the feature.
You can query it by 

`increment = cam.ExposureTime.Inc`

HughLeoWu, 2018-10-29:
Hi @thiesmoeller,

Got it, I query this parameter "cam.ExposureTime.Inc" is 35.0 
I can design my unittest to check this value, thanks.

closed 2018-10-29 by HughLeoWu