basler/pypylon #54: Getting thread Issue while running the python program

HackersSpirit, 2018-11-19:
I have written a program to extract frame form camera using  capture.GrabOne and also by continuously grabbing frame from the thread.
My problem is that many time my program terminate giving error of that unable to execute thread when other thread is running.
I am not posting the code, can post it if required but i just wanted to get an example of thread safe program in python.

thiesmoeller, 2018-11-26:
pylon and also pypylon already have a very efficient background grab thread.
[sample code](https://github.com/basler/pypylon/blob/master/samples/grabusinggrabloopthread.py)

you have to provide callback handlers, that will be called on different events [ if you install full pylon sdk the architecture is described in the C++ SDK documentation ]

Due to the GIL it is more efficient to use the pylon thread, than recreating it from python side.

thiesmoeller, 2020-11-30:
the memory corruption when using some threading in pylon has been fixed long ago

closed 2020-11-30 by thiesmoeller