basler/pypylon #56: Capture Image timeout exception

HughLeoWu, 2018-11-26:
Hi Sir.

I encounter a problem, here is my python sample codeflow:
  #camera init: 
    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
    camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
    converter = pylon.ImageFormatConverter()
    # converting to opencv bgr format
    converter.OutputPixelFormat = pylon.PixelType_BGR8packed
    converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

  while 1: #while loop
     try:
       grabresult = self.camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
       if grabresult.GrabSucceeded():
         # Access the image data
         image = self.__converter.Convert(grabresult)
         frame = image.GetArray()
       exposure_time = self.camera.ExposureTime.GetValue()
       camera_gain = self.camera.Gain.GetValue()
       grabresult.Release()
     except Exception as err:
        print('Camera exception')
        logging.info('Camera exception %s', err)
        self.camera = None

Above the sample code, it can run and no exception happen for a long time.
But I still got a issue when it run about 1 week. 

the issue description: 
The camera preview no update, and after about 1min, 

the issue log is:
"Camera exception Read operation on device '\\?\usb#vid_2676&pid_ba02&mi_00#6&20cac0a2&0&0000#{79bfeeca-947f-11e1-bc58-001b2163ef96}' failed: 'The requested operation timed out.' : TimeoutException thrown (file 'PylonUsbDevice.cpp', line 750)"

The camera product number: acA3800-14uc

How can I fix or explain this issue?
Does it possible occur over heat?

thiesmoeller, 2018-12-07:
The error you see, is probably not connected to the pypylon interface itself.
Is the camera running in PylonViewer on your system without errors?
If not, there are settings for the Basler USB cameras which could adjust to get optimal performance on your system.

thiesmoeller, 2020-11-30:
no feedback since two years

closed 2020-11-30 by thiesmoeller