basler/pypylon #57: Camera crashes after acquiring 14 images

dboonz, 2018-11-29:
Hi there,
Thanks for making this awesome wrapper! However, my cameras are not very reliable. To be precise, after fourteen frames the camera creates all kind of weird error messages and crashes.

I am using the code below to connect to three cameras (which I have to control in my experiment). What I'd like to do is to have access to all three of them individually. Grabbing a single image works, but when I get images within the last for loop, it crashes after fourteen images. Afterwards, when waiting a couple of seconds it sometimes works again for a few images but usually not.

The error message is: 
```
Traceback (most recent call last):
  File "camera_connector.py", line 71, in <module>
    im1 = grab_image(cam)
  File "camera_connector.py", line 42, in grab_image
    result = result.GetArray()
  File "C:\Users\Fibre_Lab_2\Anaconda3\lib\site-packages\pypylon\pylon.py", line 150, in func_wrapper
    return func(*args, **kwargs)
  File "C:\Users\Fibre_Lab_2\Anaconda3\lib\site-packages\pypylon\pylon.py", line 1599, in GetArray
    shape, dtype, format = self.GetImageFormat(pt)
  File "C:\Users\Fibre_Lab_2\Anaconda3\lib\site-packages\pypylon\pylon.py", line 150, in func_wrapper
    return func(*args, **kwargs)
  File "C:\Users\Fibre_Lab_2\Anaconda3\lib\site-packages\pypylon\pylon.py", line 1581, in GetImageFormat
    raise ValueError("Pixel format currently not supported")
ValueError: Pixel format currently not supported
```
Which does not really make sense as it did work for the first 14 images...

Is this a bug in pypylon or am I doing something wrong?

cheers,
Dirk

```python
import os

os.environ["PYLON_CAMEMU"] = "3"

from pypylon import genicam
from pypylon import pylon
import sys
import time
import pyqtgraph as pg

def connect_to_cameras(*, max_cameras=3):
    """ Connects to all cameras and returns a dict with all cameras identified by friendly name and the camera array. """
    # Get the transport layer factory.
    tlFactory = pylon.TlFactory.GetInstance()

    # Get all attached devices and exit application if no device is found.
    devices = tlFactory.EnumerateDevices()
    if len(devices) == 0:
        raise pylon.RUNTIME_EXCEPTION("No camera present.")

    # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
    cameras = pylon.InstantCameraArray(min(len(devices), max_cameras))

    l = cameras.GetSize()

    # Create and attach all Pylon Devices.
    result = {}
    for i, cam in enumerate(cameras):
        cam.Attach(tlFactory.CreateDevice(devices[i]))
        # Print the model name of the camera.
        print("Using device ", cam.GetDeviceInfo().GetModelName(), end='')
        friendlyname =  cam.GetDeviceInfo().GetFriendlyName()
        print(', friendly name: ',friendlyname)
        result[friendlyname] = cam
    return result, cameras

def grab_image(camera, imv=None):
    if not camera.IsGrabbing():
        camera.StartGrabbing()
    result = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
    result = result.GetArray()
    if imv is not None:
        imv.setImage(result)
        pg.QtGui.QApplication.processEvents()
    
    return result

if __name__ == '__main__':
    # connect to the cameras. As one of them often fails attempt to do it untill it works
    stop=False
    while not stop:
        try:
            camdict, camarray = connect_to_cameras(max_cameras=3)
            stop = True
        except pylon.RuntimeException:
            print("Failed to connect, trying again")
            time.sleep(1)
            
    # This should connect to three cameras
    cam_ref, cam_S, cam_P = [camdict[k] for k in camdict.keys()]
    
    cam = cam_S
    # now start grabbing a single image from cam_S, this still works
    im1 = grab_image(cam)
    
    # now start grabbing a hundred images from the camera, this fails after 14 images on cam_S, on the other cameras it works some times and some times it does not
    
    for i in range(100):
        print('frame %d of %d\r' % (i, 100))
        im1 = grab_image(cam)
        print(im1[0][0])
``

thiesmoeller, 2018-11-30:
good to hear, that you like the binding :-)

As the sample you provided does not trigger any error [ Win64/Linux64] with the latest pypylon version, here, we have some tips to trace the issue down:

RetrieveImage could fail [e.g. waiting for image where exposuretime is largen than timeout]. But there is good error reporting built in.

Please check with the [grab.py](https://github.com/basler/pypylon/blob/master/samples/grab.py) sample to learn how to use 
the 

```
GrabSucceeded
ErrorCode
ErrorDescription
```
APIs

If you give operate on a result-object from a failed acquisiton you might see further errors.
First step is always check that GrabSucceeded is True

dboonz, 2018-12-19:
Thank you, that helped. One of the cameras is a bit slower for some reason, now it works when I check if the result succeeded.

Cheers!

closed 2018-12-19 by dboonz