basler/pypylon #58: Grab timed out. The acquisition is not started.

alexdonchuk, 2018-12-01:
Hi! Camera works ok in pylon viewer 
But in pypylon

Traceback (most recent call last):
  File "/home/alex/wrk/py/aurora_backend_prototype/cam.py", line 30, in record
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
  File "/home/alex/anaconda3/lib/python3.6/site-packages/pypylon/pylon.py", line 2956, in RetrieveResult
    return _pylon.InstantCamera_RetrieveResult(self, *args)
_genicam.TimeoutException: Grab timed out. The acquisition is not started. : TimeoutException thrown (file 'InstantCameraImpl.h', line 1031)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/alex/wrk/py/aurora_backend_prototype/check_cam.py", line 5, in <module>
    record(ip0)
  File "/home/alex/wrk/py/aurora_backend_prototype/cam.py", line 45, in record
    print(e.GetDescription())
AttributeError: 'TimeoutException' object has no attribute 'GetDescription'
`import sys

from pypylon import pylon, genicam

def record(ip):
    try:
        factory = pylon.TlFactory.GetInstance()
        ptl = factory.CreateTl('BaslerGigE')
        empty_camera_info = ptl.CreateDeviceInfo()
        empty_camera_info.SetPropertyValue('IpAddress', ip)
        camera_device = factory.CreateDevice(empty_camera_info)
        camera = pylon.InstantCamera(camera_device)
        # Print the model name of the camera.
        print("Using device ", camera.GetDeviceInfo().GetModelName())

        # The parameter MaxNumBuffer can be used to control the count of buffers
        # allocated for grabbing. The default value of this parameter is 10.
        camera.MaxNumBuffer = 5

        # Start the grabbing of c_countOfImagesToGrab images.
        # The camera device is parameterized with a default configuration which
        # sets up free-running continuous acquisition.
        camera.StartGrabbingMax(10)

        # Camera.StopGrabbing() is called automatically by the RetrieveResult() method
        # when c_countOfImagesToGrab images have been retrieved.
        while camera.IsGrabbing():
            # Wait for an image and then retrieve it. A timeout of 5000 ms is used.
            grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

            # Image grabbed successfully?
            if grabResult.GrabSucceeded():
                # Access the image data.
                print("SizeX: ", grabResult.Width)
                print("SizeY: ", grabResult.Height)
                img = grabResult.Array
                print("Gray value of first pixel: ", img[0, 0])
            else:
                print("Error: ", grabResult.ErrorCode, grabResult.ErrorDescription)
            grabResult.Release()
    except genicam.GenericException as e:
        # Error handling.
        print("An exception occurred.")
        print(e.GetDescription())
        exitCode = 1

    sys.exit(exitCode)
`

adragoset, 2019-01-23:
This is a real issue. I've done some troubleshooting on ubuntu 18.01 and I've found that attempting to connect and use a camera by specifying IP is broken and will timeout on Retrieve result calls. This is true for both the python and c++ api. Ive had to roll all of my code back to grabbing the first device with a matching serial number. Wireshark caps show the camera attempting to establish a udp session back to the host on a 0.0.0.0:random port which will not work since 0.0.0.0 isnt a real ip.

rwalle, 2019-03-12:
I was seeing the same error code with my program (although it may not be due to the same cause). I upgraded pylon to 5.2, upgraded the pypylon packages to the latest 1.4.0rc1, and the problem went away. You may want to give that a try.

lcflorindo, 2020-01-09:
Hi! I'm having the same "grab timed out" issue, but with different conditions of network.
I'm using an AWS Server SUSE 12, with jumbo enables (9000 mtu) and this ping status:
`66 bytes from 100.165.10.124: icmp_seq=1 ttl=245 time=132 ms`
Try with 1500 bytes:
`From 100.45.65.1 icmp_seq=1 Frag needed and DF set (mtu = 1500)`
And 9000 bytes it doesn't even find the camera.

The code that I'm using, allows me to find the camera but depending on the PS or IPD I set, it takes more/less time to retrieve the error/time out.
```
from pypylon import pylon
from pypylon import genicam
import sys

countOfImagesToGrab = 1

try:
    ip_address = '10.12.102.124'
    factory = pylon.TlFactory.GetInstance()
    ptl = factory.CreateTl('BaslerGigE')
    empty_camera_info = ptl.CreateDeviceInfo()
    empty_camera_info.SetPropertyValue('IpAddress', ip_address)
    camera_device = factory.CreateDevice(empty_camera_info)
    camera = pylon.InstantCamera(camera_device)
    
    camera.Open()
    camera.PixelFormat.SetValue("BayerRG8")
    camera.GevSCPSPacketSize.SetValue(9000)
    camera.GevSCPD.SetValue(1000)
    print("Using device ", camera.GetDeviceInfo().GetModelName(), camera.PixelFormat())
    
    new_width = camera.Width.GetValue() - camera.Width.GetInc()
    if new_width >= camera.Width.GetMin():
        camera.Width.SetValue(new_width)
        
    camera.MaxNumBuffer = 1
    camera.StartGrabbing(countOfImagesToGrab)
    while camera.IsGrabbing():
        # Wait for an image and then retrieve it. A timeout of 5000 ms is used.
        grabResult = camera.RetrieveResult(150, pylon.TimeoutHandling_ThrowException)
```

Then the error:

```
Using device  acA1600-60gc BayerRG8
An exception occurred.
---------------------------------------------------------------------------
TimeoutException                          Traceback (most recent call last)
<ipython-input-3-509b81378fc2> in <module>
     29         # Wait for an image and then retrieve it. A timeout of 5000 ms is used.
---> 30         grabResult = camera.RetrieveResult(150, pylon.TimeoutHandling_ThrowException)
     31 

~/anaconda3/envs/pycaml8/lib/python3.6/site-packages/pypylon/pylon.py in RetrieveResult(self, *args)
   3587         """
-> 3588         return _pylon.InstantCamera_RetrieveResult(self, *args)
   3589 

TimeoutException: Grab timed out. The acquisition is not started. : TimeoutException thrown (file 'InstantCameraImpl.h', line 1031)
```

Any clue about how to solve it? I tried other versions of pypylon but didn't change anything, that's why I'm thinking about network settings...

nmarc, 2020-01-09:
Is your Network Card official supported? I had some problems with very
cheap network cards.

lcflorindo, 2020-01-15:
> Is your Network Card official supported? I had some problems with very cheap network cards. lcflorindo <notifications@github.com> schrieb am Do., 9. Jan. 2020, 19:40:
> […](#)

I guess, cause I'm working with a industrial hardware... Also I do have the same application working on a localhost perfectly. The problem is between my AWS server and my local network.

Suddenly I started to think about Pylon version or Pypylon version... There is any relation if I installed Pylon in a wrong/not perfect way or even a bug on pypylon version?

thiesmoeller, 2020-01-15:
Is inbound traffic activated in AWS?
https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/ec2-security-groups.html#default-security-group

The control channel works because per AWS default all outbound traffic and replies to this traffic are allowed.

The video stream will be sent to a port in the dynamic port range  0x8000-0xffff.
The port gets allocated by the gige transport layer.

thiesmoeller, 2025-12-15:
Closing as answered. The original issue was resolved by the user. For similar timeout issues, solutions include:
- Network configuration (AWS security groups need to allow inbound UDP traffic for video stream)
- Upgrading pylon/pypylon versions
- Using CreateFirstDevice() as a workaround when connecting by IP
- Network card compatibility issues
The root cause is typically network configuration rather than a pypylon bug.

closed 2025-12-15 by thiesmoeller