basler/pypylon #64: No grab result data is referenced. Cannot access NULL pointer.

oak-tree, 2018-12-19:
I cannot grab frame from separate python thread. I'm getting the following Error

## Error
>a.GrabSucceeded()
Traceback (most recent call last):
  File "C:\Users\oak\Anaconda2\envs\py36\lib\site-packages\IPython\core\interactiveshell.py", line 2862, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-6-75a2f62ddce8>", line 1, in <module>
    a.GrabSucceeded()
  File "C:\Users\oak\Anaconda2\envs\py36\lib\site-packages\pypylon\pylon.py", line 1719, in GrabSucceeded
    return _pylon.GrabResult_GrabSucceeded(self)
_genicam.RuntimeException: No grab result data is referenced. Cannot access NULL pointer. : RuntimeException thrown (file 'grabresultptr.cpp', line 84)

## Code
```
  def start(self):
        # start the thread to read frames from the video stream

        t = Thread(target=self.update, name=self.name, args=(self.frames,))
        t.daemon = True
        t.start()
        return self
```
```
def update(self, frames):
    # keep looping infinitely until the thread is stopped
    while True:
        # if the thread indicator variable is set, stop the thread
        if self.stopped:
            return

            grabResult = self.camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
            if grabResult.GrabSucceeded():  ## <---------------------- this throw the error
                image = self.converter.Convert(grabResult)
                # do something with image.GetArray()

            grabResult.Release()
```

Any idea?
Thanks

oak-tree, 2018-12-19:
Forgot to call 
```
        self.camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
```
It works just fine !
Thanks

closed 2018-12-19 by oak-tree

mes20a42, 2020-12-03:
Hey @oak-tree, thanks for your kind solution. May I ask where can I find the documentation which you used to resolve your issue? So far I've only been referring to pylon.py and some online samples to realize triggering using python script but there's still no successful attempt yet. Thanks in advance!

thiesmoeller, 2020-12-03:
To understand the basic concepts please refer to the c++ documentation.
Pypylon is modeled nearly 1:1 to the c++ api

mes20a42, 2020-12-03:
@thiesmoeller hmm okay I understand and solved the problem that I've raised before. Now is the index time from capturing image to my pc. I'm getting around 0.15-0.19s with lighting control in place which is extremely slow compared to continuous switching on the light and grab the image which is around 0.02s. 
May I ask if you have any suggestions to improve the speed if I would like to grab the image and flash the light once upon calling the command? Thanks in advance