basler/pypylon #65: Programming guide

AshleyRoth, 2018-12-23:
Hello! I just can not find a programming guide. Can you advise what? I have a GigE camera, I work on Linux. I want to get images and video from the camera using OpenCV, but I can’t understand how to work with Pylon. I read that you first need to get data from the camera using Pylon, then convert to OpenCV format. Thank you in advance!

pault587, 2018-12-23:
With Pylon you need a Basler GigE camera or you need to get special permission from them and new license for another brand. If you have a Basler camera then look at samples there is one in that is for OpenCV.  Are working with Pypylon for Python or with Pylon? There are also testing tools in the Code repository. I found it easiest to use the Pylon app to first view the camera and set the IP before trying to use it in code.

AshleyRoth, 2018-12-24:
@pault587. Thanks for your answer! I'm working on C and C++. I will configure camera IP with pylon configurator. I can not understand how to make a program in the C language to capture frames or video from the camera. All that has been learned so far is that you need to work with Pylon and then convert it to Opencv.

pault587, 2018-12-24:
I believe with the SDK you get a manual (something like index.html). You can download the SDK / viewer from Basler website (https://www.baslerweb.com/en/products/software/pylon-linux-x86/). In the manual there are examples of Pylon to OpenCV that you can copy and paste.  In this attached pdf there are also instructions. Pypylon here is the Python wrapper for Pylon. I normally find the camera in the viewer / ip configurator then use the example code and I have had success. I am pretty sure the examples in the manual are the same as those that are installed in the directory. I don't have access to the Linux box, it is at work, or I would paste the example here but it should be in the directory where you installed Pylon. Maybe this pdf will help...
https://www.baslerweb.com/fp-1476182890/media/downloads/documents/application_notes/AW00136801000_Getting_Started_with_pylon5_and_OpenCV.pdf  - location of the attached pdf

[AW00136801000_Getting_Started_with_pylon5_and_OpenCV.pdf](https://github.com/basler/pypylon/files/2707546/AW00136801000_Getting_Started_with_pylon5_and_OpenCV.pdf)

I hope everything above is accurate I am going from memory.

AshleyRoth, 2018-12-24:
@pault587 Thank you! I will try

closed 2020-11-30 by thiesmoeller