basler/pypylon #66: Low framerate

AshleyRoth, 2019-01-12:
Hello. Faced an unpleasant problem regarding the frame rate per second. In the characteristics of my camera it is written that the camera is issued 36 frames per second. But, for some reason, in the PylonViwer application, about 3-4 frames per second are shown only. Who can tell why?

thiesmoeller, 2019-01-12:
The effective fps or the inverse, the 'frametime' of a camera is roughly
the maximum of these main parameters:

* Readouttime ( on your camera it is less than 1/36 s )
* Exposuretime
* Transmission time ( e.g. spec fps is given for a 1000Mbit gige connection
, but you are using a 100 Mbit fast Ethernet. This will increase your
transmission time tenfold )

So please check these parameters. The Basler cameras have genicam features
to check all during operstion.
And in the manual there are sections which describe these fundamental
dependencies in detail

AshleyRoth, 2019-01-12:
@thiesmoeller Thanks! I will try it!
I would like to ask about custom permission. The question is already on the code. Now, by default, I get an image from the camera with a resolution of 2048x1536, how can I write other values in the code, for example 1280x1024? You can prompt on this issue. Very grateful to you!

pault587, 2019-01-12:
On the FPS I agree if you have a network device of 100Mbs instead of 1Gbs you will have 1/10 the rate sounds like the problem to me. 
You can change the AOI in the viewer and the camera will stay that way until you power it down.  This is useful for testing.  While you are in the viewer you can look at the **help window** for the width and height.  It tells you the name of the parameter to set for the code.  This is trick is useful for all the camera parameters.  To center the image you may also want the x and y offset, also in the viewer. It can be different between the languages are you using pypylon and Python?

AshleyRoth, 2019-01-12:
@pault587 Thanks! About languages, i using Pylon with c++

AshleyRoth, 2019-01-20:
@pault587 @thiesmoeller hello again! I tried to use a network card with 1Gbs, the situation has not changed. The same small FPS, at the moment I use a POE adapter to connect the camera, in the characteristics of the POE adapter it is written that the speed is 10/100 / 1000mbs, but on the computer the connection still shows 100mbs. What could be the problem? All the same because of POE?

Regards!

pault587, 2019-01-20:
In the pylon viewer it will show you the fps and bandwidth while you stream.  If you are still under 100Mbs I would try another PoE adapter but first verify you PC network card.  Also check in the viewer that the fps is set to auto or at a high frame rate.  If you want to check your bandwidth manually, multiple your image fps, height, width, channels and 8 (assuming 8 bit camera) divide by 10^6 for mega Bytes.
To check your PC card;
If you are on linux you can open a terminal and type "ethtool eth0" or "ethtool (your nic adapter name)" to verify it is 1000Mb/s.  If you don't know your adapter name type "ifconfig" in the terminal and it will be printed on the left side.
For Windows go to settings and network connections, change adapter options, double click you nic card I believe it will say 1.0Gbps

AshleyRoth, 2019-01-20:
@pault587
Thanks for answer!
About network card, when I connect to Ethernet in the options i see 1000mbs, but in the camera case I see in options 100mbs... so then i try change my Poe adapter. 
I will check and back again

AshleyRoth, 2019-01-21:
And again hello! @pault587 
Unfortunately, I don’t know where to ask my questions, I found only this git. In general, perhaps you can help me, at the moment I have another question. Even two questions.
- First, what is responsible for "Packet Size" in the PylonViewer settings, what is the difference between 1500 and 8192, for example?
- Second. I wrote code here to get frames using OpenCV, but I ran into two problems. 1 problem, I can not do cvReleaseImage, the program crashes if you use cvReleaseImage. The second problem, after some period of time, the program crashes with the error "terminate called after throwing an instance of 'genicam_3_1_basler_pylon_b5_1 ..". I attach the program code:
`	IplImage* pImgEmbed;
	int saveImg = 1;
	int grabbedImages = 0;
	CPylonImage resultCV;
	int counter;
	char path[255];
   	 char num[10];
  	char jpg[10] = ".jpg";
	int channels = 3;
	unsigned char bytes_array[2048*1536*3];

int main(int argc, char* argv[])
{
    int exitCode = 0;
    // Before using any pylon methods, the pylon runtime must be initialized. 
    PylonInitialize();

   // try
   // {
				CImageFormatConverter converter; 
				converter.OutputPixelFormat = PixelType_RGB8packed;

				// This smart pointer will receive the grab result data.                
				CInstantCamera Camera( CTlFactory::GetInstance().CreateFirstDevice());
				cout << "Using device " << Camera.GetDeviceInfo().GetModelName() << endl; 

				Camera.Open();
				CIntegerPtr width( Camera.GetNodeMap().GetNode( "Width"));
				CIntegerPtr height( Camera.GetNodeMap().GetNode( "Height"));
				CEnumerationPtr pixelFormat( Camera.GetNodeMap().GetNode( "PixelFormat"));
				
				//Camera.StartGrabbing( c_countOfImagesToGrab, GrabStrategy_LatestImages);
				Camera.StartGrabbing( GrabStrategy_OneByOne);
				CGrabResultPtr ptrGrabResult;
				
				
		
					while (Camera.IsGrabbing()) {
						Camera.RetrieveResult(5000, ptrGrabResult, TimeoutHandling_ThrowException);
						cout << "grab working" << endl;

						int width = ptrGrabResult->GetWidth();
						int height = ptrGrabResult->GetHeight();
						cout << "SizeX: " << width << endl;
						cout << "SizeY: " << height << endl;

						converter.Convert(resultCV, ptrGrabResult);
						pImgEmbed=cvCreateImage(cvSize(resultCV.GetWidth(), resultCV.GetHeight()), IPL_DEPTH_8U, 3);
						
						cvSetData(pImgEmbed, (uint8_t *) resultCV.GetBuffer(), pImgEmbed->width*3);
						
						//cvReleaseImage(&pImgEmbed);  //if uncomment app is drop when running
						

						if (saveImg == 1) {
							strcpy(path, "images/pic");
							sprintf(num, "%03i", counter);
							strcat(path, num);   
							strcat(path, jpg);
							printf("Saving: %s\n", path);
							cvSaveImage(path, pImgEmbed);
							counter++;
						}

					}
				
				
				return 0;
         
    PylonTerminate(); 

    return exitCode;
}`

pault587, 2019-01-21:
On your first question, I don’t know your operating system so I’ll let you try and search for enabling jumbo frames or jumbo packets.

AshleyRoth, 2019-01-21:
@pault587 Oh sorry, my OS is Ubuntu 16.04. What about second question? 

Regards

thiesmoeller, 2020-11-30:
deleted issue

closed 2020-11-30 by thiesmoeller