basler/pypylon #67: question: grab HDR image sequence

ZacDiggum, 2019-01-21:
Hi there,

I need to grab a sequence of images with different exposure times (for HDR imaging) as fast as possible. What is the right way to do this? Maybe someone could give a short code example? Thanks in advance...

thiesmoeller, 2019-01-21:
Hi,

What camera model do you use?
E.g. the basler ace supports exposure bracketing with the sequencer feature

ZacDiggum, 2019-01-22:
Hi @thiesmoeller ,
I'm using a [acA3088-57um](https://www.baslerweb.com/de/produkte/kameras/flaechenkameras/ace/aca3088-57um/).
Best would be a short code example to get started. I'd like to implement sort of 'real-time HDR' which means I want to grab a series of images with defined exposure times, process them into a tonemapped HDR image, display it and then repeat.
Your help is very much appreciated, thank you.

thiesmoeller, 2019-01-23:
Hi @ZacDiggum,

here is a link to the official documentation of the sequencer feature for the USB3Vision camera:

[Basler Docs: Sequencer](https://docs.baslerweb.com/#t=en%2Fsequencer_(usb_cameras).htm&rhsearch=sequencer&rhhlterm=sequencer)

will provide a stripped down version with your exposure bracketing use-case in the next days

thiesmoeller, 2019-01-25:
# Exposure bracketing using the sequencer feature of USB3 ace devices

```python
import pypylon.pylon as py
import numpy as np
```

```python
# open the first USB device
info = py.DeviceInfo()
info.SetDeviceClass("BaslerUsb")
cam = py.InstantCamera(py.TlFactory.GetInstance().CreateFirstDevice(info))
```

```python
# this code only works for ace USB
if not cam.GetDeviceInfo().GetModelName().startswith("acA"):
    print("_This_ sequencer configuration only works to basler ace USB")
```

```python
# open device
cam.Open()
```

## setup camera.

sequencer mode is not running if the auto-functions are active

```python
cam.UserSetSelector = "Default"
cam.UserSetLoad.Execute()
```

```python
# mean exposure time 20ms
exp_0 = 2000
exp_1 = 20000
exp_2 = 200000
```

### activate  chunks [ embedded data ]
This allows to read the exposuretime an image has been taken with from the image data

```python
# enable camera chunk mode
cam.ChunkModeActive = True
# enable exposuretime chunk
cam.ChunkSelector = "ExposureTime"
cam.ChunkEnable = True
```

### sequencer setup

```python
cam.SequencerMode = "Off"
cam.SequencerConfigurationMode = "On"
```

### setup set 0

```python
cam.SequencerSetSelector = 0

cam.ExposureTime = exp_0

cam.SequencerSetSave.Execute();
```

### setup set 1

```python
cam.SequencerSetSelector = 1

cam.ExposureTime = exp_1

cam.SequencerSetSave.Execute();
```

### setup set 2

```python
cam.SequencerSetSelector = 2

cam.ExposureTime = exp_2

# select that we jump to set 0 after this set
# path 1 is the _next_ path / path 0 would be _reset_ path
cam.SequencerPathSelector = 1
cam.SequencerSetNext = 0

cam.SequencerSetSave.Execute();

```

### enable sequencer

```python
cam.SequencerConfigurationMode = "Off"

# this will set the first sequencer set to set _0_ as side effect
cam.SequencerMode = "On"
```

### test capture with enabled sequencer mode

```python
# grab 4 sets of 3 images
cam.StartGrabbingMax( 4 * 3)

while cam.IsGrabbing():
    res = cam.RetrieveResult(1000)
    exp_time_chunk = res.ChunkDataNodeMap.ChunkExposureTime.Value
    print("%d\t%6.0f\t%6.2f" %(res.BlockID, exp_time_chunk, np.mean(res.Array)))
    # ... do something with the images
    
    # return to buffer queue
    res.Release()
    
cam.StopGrabbing()
```

    0	  2000	 11.81
    1	 20000	 58.71
    2	200000	125.30
    3	  2000	 11.80
    4	 20000	 58.71
    5	200000	125.31
    6	  2000	 11.80
    7	 20000	 58.71
    8	200000	125.32
    9	  2000	 11.81
    10	 20000	 58.71
    11	200000	125.32

### Shutdown session after acquisition

This is required to rerun the notebook

```python
cam.SequencerMode = "Off"
```

```python
cam.Close()
```

ZacDiggum, 2019-01-26:
Thank you very much, @thiesmoeller ! I'll look into it and adapt your example to my use case.

ZacDiggum, 2019-01-30:
@thiesmoeller 
I think it would be a good idea to include your sequencer example into the samples folder of pypylon. Thanks again for your help. I'll close this issue.

closed 2019-01-30 by ZacDiggum

Filip2002, 2019-07-02:
Does anybody knows what do I have to change in this thiesmoeller's code to make it work on my basler gigE camera ( I am using model acA1300-60gc)?  
I suppose that I have to change the first part, when I create camera in python, but I couldn't find how to open gigE camera using pypylon.  

> # Exposure bracketing using the sequencer feature of USB3 ace devices
> ```python
> import pypylon.pylon as py
> import numpy as np
> ```
>
> ```python
> # open the first USB device
> info = py.DeviceInfo() [...]