basler/pypylon #70: basler camera call using pypylon issue

Thyagutamizhan, 2019-01-23:
while callling the basler camera using pypylon i faced an issue like,

Process finished with exit code -1066598274 (0xC06D007E)

Thyagutamizhan, 2019-01-24:
script used:
import pypylon
from pypylon import pylon

cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
cam.Open()
print(cam.GetDeviceInfo)
cam.Close()

print("done")

os: windows 10 64 bit
camera: Basler acA640
coneected via usb 3.0

thiesmoeller, 2019-01-24:
this could be an uncaught exception. Please test again with this script:

```
from pypylon import pylon

import sys
from distutils.util import get_platform

print(sys.version, platform.os.name )
print(get_platform())

try:
    cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
    cam.Open()
    print(cam.GetDeviceInfo().GetFullName())
    cam.Close()
    print("done")
except Exception as e:
    print("Error")
    print(e)
    

```

Thyagutamizhan, 2019-01-25:
it gives a result like this:

C:\Users\Dell\python.exe C:/Users/Dell/PycharmProjects/open_cv/test.py
3.7.2 (tags/v3.7.2:9a3ffc0492, Dec 23 2018, 23:09:28) [MSC v.1916 64 bit (AMD64)]
win-amd64

Process finished with exit code -1066598274 (0xC06D007E)

thiesmoeller, 2019-01-25:
Hi @Thyagutamizhan, 
did you compile the pypylon version on your own?

In this case you should first run the internal testsuites of pypylon to make sure, that everything internally 
is working well.

We haven't yet tested and released against python 3.7.2

Thyagutamizhan, 2019-01-25:
yeah thiesmoeller,

Actually im a beginner,I dont know how to fix this issue. lemme get suggestion for perfect version of python.

thiesmoeller, 2019-01-25:
hi @Thyagutamizhan ,

escpecially as a beginner [ welcome :-)  ] it is best to choose a tested combination of python/pypylon.

For the python version choose:
the **3.6.8** from [python.org/downloads](https://www.python.org/downloads/windows/)

Then you can simply use the command i mentioned

`pip install pypylon`

As a hint:
 make sure, that you really remove the python 3.7 environment you added, otherwise you might run  [ depending on your path settings ]  into some issues....

Thyagutamizhan, 2019-01-25:
Hi thiesmoeller,
   Actually the issue has been fixed, thanku for that. I need to know why its not working in python 3.7.2 version.  Kindly lemme know if there is any tutorials for pypylon to work with pylon sdk.

thiesmoeller, 2019-01-25:
As pypylon is a SWIG generated extension, you have to use a C++ debugger to understand what is not running on your platform.
-> at the moment I would recommend to stay with 3.6 as long the 3.7 is available.

pypylon is modeled very close to the pylon C++ SDK.
So it is good install the full pylon c++ sdk from www.baslerweb.com and read the docs regarding the architecture.

Then move forward with playing with the scripts in the samples folder of this project.

Thyagutamizhan, 2019-01-28:
ok thiesmoeller, thank you.

closed 2019-02-20 by Thyagutamizhan

amarnathreddy0201, 2022-01-07:
Hi Sir,
I want to connect to cameras how can i use at a time.
cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice()) This is for first camera .
Please help me regarding this.

Thanks,
Amar Reddy S