basler/pypylon #75: IssueScheduledActionCommand not available

theSaLi, 2019-01-28:
Hello everybody,

I have 3 acA3088-16gc cameras on a Frame Grabbers / Video Capture Cards PCIe-GIE72/74 from Adlink  and I would like to acquire images from 3 cameras at the same time. 

I followed the instruction from scheduled action command (https://docs.baslerweb.com/#t=en%2Fscheduled_action_commands.htm#Issuing_a_Scheduled_Action_Command) and tried to translate C++ code into python. But couldn't find the IssueActionCommand() function anywhere. Could anyone please tell me how to create a GigeTL and use the IssueActionCommand(). 

Another example in C++ can be found in sample code Grab_UsingActionCommand.cpp in C++ Programmer's Guide and Reference.

If the IssueActionCommand() is not implemented in pypylon yet, could anyone please tell me how to let 3 cameras acquire images at the same time with pypylon?

Thank you very much for your help!

thiesmoeller, 2019-01-30:
The GigeTL ( nor the other TLs specific versions)  is currently not mapped to python.

As sending an action command out doesn't require a connection to a camera, a quick workaround would be to use e.g. ctypes to call into PylonC and send the action command aside from pypylon.

we are currently looking into the best way to modify the TLFactory to support the specifc TL interfaces

thiesmoeller, 2020-11-30:
access to gig tl is now possible with current pypylon version

closed 2020-11-30 by thiesmoeller