basler/pypylon #76: how to grab image from camera using external trigger?

Thyagutamizhan, 2019-01-29:


aturpin88, 2019-01-30:
First, check what type of trigger do you need (either hardware or software). There is an option in the pylon menu. Open your camera, go to: 
Acquisition Control -> Trigger Mode -> On
in Acquisition Control, Trigger Source selects the type of triggering that you need. If you want hardware trigger, check the specs of your camera and select a line. If software, select "Software". 

I hope this helps.

thiesmoeller, 2019-01-30:
For a basler ace gige:
[basler product documentation: Configuring a Hardware Trigger Source](https://docs.baslerweb.com/#t=en%2Ftrigger_source.htm&rhsearch=hardware%20trigger)

really can recommend to jump into the product documentation. 
pypylon only provides access to the features of the camera.
The features itself are documented in the basler product documentation. 

[docs.baslerweb.com](https://docs.baslerweb.com)

aturpin88, 2019-02-02:
I see your point. What I usually do is first save the configuration file in Pylon Viewer with all the camera features I want to (exposure, frame size, triggering, etc). Then I load it directly in Python: 
```
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
folder = "folder_dir_of_your_config_file"
camera.Open()
pylon.FeaturePersistence.Load(folder+"acA640-750um_22903738.pfs", camera.GetNodeMap()) # Config file with the capturing properties
```
Does this help?

Thyagutamizhan, 2019-02-05:
Actually i just wanna configure a hardware trigger and capture a image using the hardware trigger and access that image using pypylon. How can i do that ? Is there any sample codes are tutorials available for pypylon.?

thiesmoeller, 2019-02-05:
Hi @Thyagutamizhan,

to reach the highest performance you should work along [grabusing_grabloop](https://github.com/basler/pypylon/blob/master/samples/grabusinggrabloopthread.py) programming sample.

Of course skipping all those softwaretrigger configurations and exchange them with your specific HW trigger configuration. [ the exact sequence depends on your specifc use case and the camera interface USB3/GigE/Cameralink/LVDS/MIPI ... it is documented in the manual ]

The important point is, that for a HW trigger you don't know when the image will arive. Thus it's getting tricky to use all functions, that have a timeout parameter ( like RetrieveResult ).

Using the pylon internal grab-thread, will automatically call the OnImage Callback when the image arrives.
In this thread you can do your processing with a very low latency to the point in time, when the image is captured in your machine.

Thyagutamizhan, 2019-02-08:
thanks thiesmoeller,
  Here im coming to the point, how to set the hardware configuration and where should i set the camera paramerters. I already asked this only, i think so i should get a perfect answer.

thiesmoeller, 2019-02-08:
@Thyagutamizhan : for the configuration look into the camera documentation that has been already mentioned. pypylon is the generic API for basler cameras. The features and values belong to the camera itself. ( [hardware trigger in docs.baslerweb.com](https://docs.baslerweb.com/#t=en%2Ftriggered_image_acquisition.htm&rhsearch=hardware%20trigger&rhhlterm=hardware%20trigger) )

If your question is still: "how to configure a basler camera using pypylon"
in the `samples` folder there are multiple example to configure the camera.

```
camera = pylon.InstantCamera(
    pylon.TlFactory.GetInstance().CreateFirstDevice())

camera.Open()

camera.Gain = 42.0
camera.TriggerSelector = "FrameStart"
camera.TriggerMode = "On"

...
```

Thyagutamizhan, 2019-02-08:
import logging
import config
from pypylon import pylon
import os
import numpy as np

COLLECT_PATH = '/data'
formatter = logging.Formatter( '%(asctime)s:%(levelname)s:%(process)d:%(thread)d:%(module)s:%(funcName)s:%(lineno)s:%(message)s')
logger = logging.getLogger(__name__)

async def callback(img, camera_model_name, img_id, img_timestamp):
    ''' callback gets called from imaging event trigger. after image grab is finished this callback is called with image array. store the image as file. '''
    cam_folder = config.camera["nir" if "nir" in camera_model_name.lower()[-3:] else "rgb"]
    img_name = f'{img_id}_{img_timestamp}'
    img_filepath = os.path.join(COLLECT_PATH, cam_folder, img_name)
    with open(img_filepath, 'wb') as f:
        np.save(f, img)

class ImageEventListener(pylon.ImageEventHandler):
    """ Contains an Image Event Handler that prints a message for each event method call. also supports callback to pass in captured image. """
    def __init__(self, callback):
        super(ImageEventListener, self).__init__()
        self.callback = callback
    def OnImagesSkipped(self, camera, countOfSkippedImages):
        logger.info("OnImagesSkipped event for device {}".format( camera.GetDeviceInfo().GetModelName()))
        logger.info("{} images have been skipped.".format(countOfSkippedImages))
    def OnImageGrabbed(self, camera, grabResult, run_coroutine=None):
        logger.info("OnImageGrabbed event for device {}".format( camera.GetDeviceInfo().GetModelName()))
        if grabResult.GrabSucceeded():
            logger.info("SizeX: {}".format(grabResult.GetWidth()))
            logger.info("SizeY: {}".format(grabResult.GetHeight()))
            img = grabResult.GetArray()
            print (grabResult.GetID(), grabResult.GetTimeStamp(), grabResult.GetTimeStamp())
            try:
              if self.callback is not None:
                 return run_coroutine(self.callback, img=img, camera_model_name=camera.GetDeviceInfo().GetModelName(), img_id=grabResult.GetID(), img_timestamp=grabResult.GetTimeStamp())
            except Exception as e:
                logger.error(e)
        else:
             logger.info("Error: {}".format(grabResult.GetErrorCode(), grabResult.GetErrorDescription()))

class ConfigurationEventListener(pylon.ConfigurationEventHandler):
    """ Contains a Configuration Event Handler that prints a message for each event method call. """
    def OnAttach(self, camera):
        logger.info(f"OnAttach event")
    def OnAttached(self, camera):
        logger.info(f"OnAttached event for device {camera.GetDeviceInfo().GetModelName()}")
    def OnOpen(self, camera):
        logger.info(f"OnOpen event for device {camera.GetDeviceInfo().GetModelName()}")
    def OnOpened(self, camera):
        logger.info(f"OnOpened event for device {camera.GetDeviceInfo().GetModelName()}")
    def OnGrabStart(self, camera):
        logger.info(f"OnGrabStart event for device {camera.GetDeviceInfo().GetModelName()}")
    def OnGrabStarted(self, camera):
        logger.info(f"OnGrabStarted event for device {camera.GetDeviceInfo().GetModelName()}")
    def OnGrabStop(self, camera):
        logger.info(f"OnGrabStop event for device {camera.GetDeviceInfo().GetModelName()}")
    def OnGrabStopped(self, camera):
        logger.info(f"OnGrabStopped event for device {camera.GetDeviceInfo().GetModelName()}")
    def OnClose(self, camera):
        logger.info(f"OnClose event for device {camera.GetDeviceInfo().GetModelName()}")
    def OnClosed(self, camera):
        logger.info(f"OnClosed event for device {camera.GetDeviceInfo().GetModelName()}")
    def OnDestroy(self, camera):
        logger.info(f"OnDestroy event for device {camera.GetDeviceInfo().GetModelName()}")
    def OnDestroyed(self, camera):
        logger.info(f"OnDestroyed event")
    def OnDetach(self, camera):
        logger.info(f"OnDetach event for device {camera.GetDeviceInfo().GetModelName()}")
    def OnDetached(self, camera):
[... 18 lines cut ...]
        camera1.GetNodeMap()'''
        camera1.RegisterConfiguration(pylon.ConfigurationEventHandler(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
        camera1.RegisterConfiguration(ConfigurationEventListener(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
        logger.info("Initalizing device : {}".format(camera1.GetDeviceInfo().GetModelName()))

if __name__ == '__main__':
    logging.basicConfig(level='DEBUG', format=formatter._fmt)
    cams = Camera()
    cm=cams.camera1
    event_listener = ImageEventListener(callback=callback)
    cm.RegisterImageEventHandler( event_listener, pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
    cm.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)

gives error like this:

Traceback (most recent call last):
  File "C:/Users/Dell/Projects/basler_samples/test1.py", line 96, in <module>
    cams = Camera()
  File "C:/Users/Dell/Projects/basler_samples/test1.py", line 84, in __init__
    pylon.FeaturePersistence.Load(folder + "acA640-750uc_22793686.pfs", camera1.GetNodeMap())
  File "C:\Program Files\Python36\lib\site-packages\pypylon\pylon.py", line 6580, in Load
    return _pylon.FeaturePersistence_Load(FileName, pNodeMap, validate)
_genicam.RuntimeException: Error while writing node 'ExposureAuto': Device '\\?\usb#vid_2676&pid_ba02&mi_00#6&262302a4&0&0000#{79bfeeca-947f-11e1-bc58-001b2163ef96}' not open when trying to write to address 0x40424; length 4. : LogicalErrorException thrown (file 'PylonUsbDevice.cpp : RuntimeException thrown (file 'FeaturePersistence.cpp', line 64)

Actually i have done the configurations in pylon viewer i wanna use that using pypylon to grab a image and process it.

thiesmoeller, 2019-02-08:
... please look at my example ... and the error messages you receive from the system!

putting a `#` in front of an expression creates a line comment.....
so 
```
...
#cameras.Open() 
...
```

will never open the camera ..... this is why you see exceptions of not beeing able to communicate with the device ....

`... Device '\?\usb#vid_2676&pid_ba02&mi_00#6&262302a4&0&0000#{79bfeeca-947f-11e1-bc58-001b2163ef96}' not open when trying to write to address
`
`Node is not writable : AccessException thrown in node 'TriggerMode'`

closed 2019-02-11 by Thyagutamizhan

cute-tiger, 2019-06-04:
Please direct which specific example I can look at for setting the hardware trigger using pypylon API?

thasnimolvs, 2021-01-15:
> > First, check what type of trigger do you need (either hardware or software). There is an option in the pylon menu. Open your camera, go to:
> > Acquisition Control -> Trigger Mode -> On
> > in Acquisition Control, Trigger Source selects the type of triggering that you need. If you want hardware trigger, check the specs of your camera and select a line. If software, select "Software".
> > I hope this helps.
> 
> yeah its helpful,, what i'm trying to ask was, is there any hardware trigger configurations available like this:
> camera.RegisterConfiguration(pylon.SoftwareTriggerConfiguration(), pylon.RegistrationMode_ReplaceAll,
> pylon.Cleanup_Delete)

am facing with same problem. how to do this and also how to change fps