basler/pypylon #80: Using more than 5 cameras

marsue, 2019-02-06:
I'm trying to capture 6 USB-cameras simultaneously. With 5 everything runs fine, but when using a 6th camera pypylon is crashing, regardless of how and where they are plugged in. If I remove one camera on any USB-Port it is working again, so I guess this is a software problem.

This minimal example
```python
#!/usr/bin/python3.6
from pypylon import pylon, genicam

tlFactory = pylon.TlFactory.GetInstance()
devices = tlFactory.EnumerateDevices()
cameraArray = pylon.InstantCameraArray(len(devices))

for i, camera in enumerate(cameraArray):
    camera.Attach(tlFactory.CreateDevice(devices[i].GetFullName()))
    camera.Open()
    camera.StartGrabbing()
```
throws the following traceback
```
Traceback (most recent call last):
  File "./minimal.py", line 9, in <module>
    camera.Attach(tlFactory.CreateDevice(devices[i].GetFullName()))
  File "/home/user/.local/lib/python3.6/site-packages/pypylon/pylon.py", line 1060, in CreateDevice
    return _pylon.TlFactory_CreateDevice(self, *args)
_genicam.RuntimeException: Failed to open device '2676:ba02:x:y:z' for XML file download. Error: 'UX Status: Libusb error: LIBUSB_ERROR_IO.' : RuntimeException thrown (file 'PylonUsbTL.cpp', line 474)

```

When removing `camera.StartGrabbing()` and adding `cameraArray.StartGrabbing()` after the for loop I get this instead
```
Traceback (most recent call last):
  File "./minimal.py", line 13, in <module>
    cameraArray.StartGrabbing()
  File "/home/user/.local/lib/python3.6/site-packages/pypylon/pylon.py", line 4062, in StartGrabbing
    return _pylon.InstantCameraArray_StartGrabbing(self, *args)
_genicam.RuntimeException: WaitObject duplicate failed (0): Too many open files : RuntimeException thrown (file 'WaitObject_Unix.cpp', line 47)
```

And finally when running the `samples/grabmultiplecameras.py` example I just get
```
terminate called after throwing an instance of 'GenICam_3_1_Basler_pylon_v5_1::RuntimeException'
```
The cameras are acA1920-40uc and I'm running on Linux.

Any help would be appreciated, as I'm stuck at this point for quite some time now.

thiesmoeller, 2019-02-07:
hi @marsue, is the 6-camera system running in pylon viewer on your linux machine?

marsue, 2019-02-07:
Yes @thiesmoeller, everything works fine in pylon viewer. I can see all 6 camera images simultaneously.

thiesmoeller, 2019-02-08:
* Problem: When trying to grab frames from multiple cameras, the error message
  "PrepareGrab (StartStreaming) failed for device" or "Insufficient system
  resources exist to complete the API" are displayed.

  Solution: This condition can occur when the number of available open file
  descriptors is exhausted. The limit can often be raised for the current
  process with the command: ulimit -n <number>

marsue, 2019-03-12:
This was indeed the case. Thank you the help.

closed 2019-03-12 by marsue

lucyd42, 2020-03-03:
Is there a limit to how many cameras can pylon handle? If i want to run 15-20 camers, is it possible? I will be using i7 and 2080ti for this.

thiesmoeller, 2020-03-04:
Even larger camera setups are not a problem for pylon.

Depending on Interface usb3 or gige you have to tune bandwidth settings accordingly.

lucyd42, 2020-03-04:
Thank you @thiesmoeller