basler/pypylon #81: request for windows 32bit / python 3.7 binary

tiggersternchen, 2019-02-08:
Could you please complile and upload the "pypylon-1.3.1-cp37-cp37m-win32.whl" version (Python 3.7, Windows x32) ?

Many thanks in advance
Sven

Knoose, 2019-03-01:
I am also looking for this version

Knoose, 2019-03-04:
I was able to build it myself. I tried to use the feature/python37 branch but that failed pretty miserably. If you need me to share it, I can but ideally Basler should release a working binary.

tiggersternchen, 2019-03-11:
I would be greatful if you could share your compiled version until Basler releases an official build.

Thanks in advance
Sven

stefanklug, 2019-03-12:
Hi,
I just tagged a 1.4.0 release candidate. The wheel is here:
https://github.com/basler/pypylon/releases/download/1.4.0rc1/pypylon-1.4.0rc1-cp37-cp37m-win32.whl
Feel free to test it.

tiggersternchen, 2019-03-14:
Thank you for your effort Stefan,

it's working for our USB3 camera. Unfortunately we have recognized that the pypylon wrapper does not support 1394 firewire cameras. Do you or somebody else know a workaround ?

Thank you very much
Sven

Knoose, 2019-03-21:
Have you looked that this? https://github.com/basler/pypylon/issues/47

thiesmoeller, 2020-11-30:
original issue is fixed

closed 2020-11-30 by thiesmoeller