basler/pypylon #83: Camera Basler inside a class?

Siset, 2019-02-14:
Hi everyone,

I am trying to declare a class to encapsulate basler cameras. I have two USB cameras acA5472-17uc. For instance, I would like to declare them once and then whenever from the main capture a pair of images. The class defined was created following the examples and it looks like:
`
class CameraBasler():
	
	def __init__(self):
		self.cameras=[]
		tlFactory = pylon.TlFactory.GetInstance()

		# Get all attached devices and exit application if no device is found.
		devices = tlFactory.EnumerateDevices()
		if len(devices) == 0:
			raise pylon.RUNTIME_EXCEPTION("No camera present.")

		# Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
		cameras = pylon.InstantCameraArray(min(len(devices), 2))

		# Create and attach all Pylon Devices.
		for i, cam in enumerate(cameras):
			cam.Attach(tlFactory.CreateDevice(devices[i]))
			cam.RegisterConfiguration(pylon.SoftwareTriggerConfiguration(), pylon.RegistrationMode_ReplaceAll,pylon.Cleanup_Delete)
			# For demonstration purposes only, add a sample configuration event handler to print out information
			# about camera use.t
			cam.RegisterConfiguration(ConfigurationEventPrinter(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
			# The image event printer serves as sample image processing.
			# When using the grab loop thread provided by the Instant Camera object, an image event handler processing the grab
			# results must be created and registered.
			cam.RegisterImageEventHandler(ImageEventPrinter(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
			# For demonstration purposes only, register another image event handler.
			cam.RegisterImageEventHandler(SampleImageEventHandler(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
		# Start the grabbing using the grab loop thread, by setting the grabLoopType parameter
		# to GrabLoop_ProvidedByInstantCamera. The grab results are delivered to the image event handlers.
		# The GrabStrategy_OneByOne default grab strategy is used.
		cameras.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)
		self.cameraL=cameras[0]
		self.cameraR=cameras[1]
		# The grabbing is stopped, the device is closed and destroyed automatically when the camera object goes out of scope.
	def takeStereo(self):
		
		if self.cameraL.WaitForFrameTriggerReady(100, pylon.TimeoutHandling_ThrowException) and self.cameraR.WaitForFrameTriggerReady(100, pylon.TimeoutHandling_ThrowException):
			self.cameraL.ExecuteSoftwareTrigger()
			self.cameraR.ExecuteSoftwareTrigger()

		#time.sleep(1)
		`
The main problem here is that after initialization, The grabbing is stopped, the device is closed and destroyed automatically when the camera object goes out of scope.

My question is that the implementation that I am trying is possible and how to do it.

Thanks in advanced,
Narcis

thiesmoeller, 2019-02-16:
good catch.

There must be an issue with the index operator in InstantCameraArray. The problem is that `self.cameraL=cameras[0]` doesn't increment the refcount of cameras. Thus there is no reference to `cameras` when leaving the constructor.

if you make `cameras`  an instance variable it works:

```
import os
os.environ["PYLON_CAMEMU"] = "2"
import pypylon.pylon as pylon
import pypylon.genicam as geni
#import from samples folder
from configurationeventprinter import ConfigurationEventPrinter
from imageeventprinter import ImageEventPrinter
import time

class CameraBasler(object):

    def __init__(self):
        tlFactory = pylon.TlFactory.GetInstance()

        # Get all attached devices and exit application if no device is found.
        devices = tlFactory.EnumerateDevices()
        if len(devices) == 0:
            raise pylon.RUNTIME_EXCEPTION("No camera present.")

        # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
        self.cameras = pylon.InstantCameraArray(min(len(devices), 2))

        # Create and attach all Pylon Devices.
        for i, cam in enumerate(self.cameras):
            cam.Attach(tlFactory.CreateDevice(devices[i]))
            cam.RegisterConfiguration(pylon.SoftwareTriggerConfiguration(), pylon.RegistrationMode_ReplaceAll,
                                      pylon.Cleanup_Delete)
            # For demonstration purposes only, add a sample configuration event handler to print out information
            # about camera use.t
            cam.RegisterConfiguration(ConfigurationEventPrinter(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
            # The image event printer serves as sample image processing.
            # When using the grab loop thread provided by the Instant Camera object, an image event handler processing the grab
            # results must be created and registered.
            cam.RegisterImageEventHandler(ImageEventPrinter(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
        # Start the grabbing using the grab loop thread, by setting the grabLoopType parameter
        # to GrabLoop_ProvidedByInstantCamera. The grab results are delivered to the image event handlers.
        # The GrabStrategy_OneByOne default grab strategy is used.
        self.cameras.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)
        self.cameraL = self.cameras[0]
        self.cameraR = self.cameras[1]

    # The grabbing is stopped, the device is closed and destroyed automatically when the camera object goes out of scope.
    def takeStereo(self):
        if self.cameraL.WaitForFrameTriggerReady(100,
                                                 pylon.TimeoutHandling_ThrowException) and self.cameraR.WaitForFrameTriggerReady(
                100, pylon.TimeoutHandling_ThrowException):
            self.cameraL.ExecuteSoftwareTrigger()
            self.cameraR.ExecuteSoftwareTrigger()

cams=CameraBasler()
for i in range(10):
    cams.takeStereo()
    time.sleep(1)
```

Siset, 2019-02-18:
Hi @thiesmoeller! 

Thank you very much for your help! Now it works flawlessly! 👍  
Really appreciate your help! 

Thanks!!!!
Narcis

closed 2019-02-18 by Siset